opencardev/crankshaft #163: Declaration of work

htruong, 2018-07-11:
If you have worked on CS development on any part (code/art/UI/patch/documentation), please declare here in this format:

```
The month of ?/20??.
--
Email (to contact about payment detail):
What you worked on (one sentence):
How much time it took you (hours):
Anything you have to purchase to get it done? 
```

For example:
```
The month of June/2018
--
Email: htruong@tnhh.net
What you worked on: Porting CS to the TinkerBoard
How much time: 20
Purchases: $0
```

Any donation I received during the same month will be reviewed on the 10th of the next month and will be distributed to all contributors accordingly to hours worked.

htruong, 2018-07-11:
```
The month of June/2018
What you worked on: Reddit support, some CS bookkeeping.
How much time: 2 hours
Purchases: $0
```

cydia2020, 2018-07-12:
Don't know whether this counts

```
The month of June/2018
--
Email: cydia2020@outlook.jp
What you worked on: Wiki Side Menu, IQAudio Sound card guide
How much time: 20 mins
Purchases: $0
```

hawkeyexp, 2018-07-13:
`The month of June/2018
--
Email: hawkeyexp@gmail.com
What you worked on: Hm whole system *lol*
How much time: hours over hours - to much to collect sorry...
Purchases: $0`

inspector71, 2018-08-19:
The month of 8/2018.
--
Email: N/A (appreciate the remunaration offer though!)
What you worked on: Wiki (though it looks like I've now been locked out?)
How much time: Probably a dozen hours all up but I'm not really counting :)
Purchase: $0

htruong, 2018-08-20:
@cydia2020 @hawkeyexp @inspector71,

Thanks for all of your contributions. I was on a leave so it's kinda late this month!

I would like to inform that we have an income of 21.85 USD in the month of 7/2018. Since this month is the first month, I'd like to propose this split:

@htruong: 1 USD
@cydia2020: 1 USD
@hawkeyexp: 19.85 USD
@inspector71: 0 USD, as requested, but with a big thank you!

If you have bitcoins or would like to receive bitcoin, I am happy to provide payment by bitcoin as well, if you'd like to (please let me know of your bitcoin address). If you want to "keep the balance," I will perhaps try to roll over the balance every month.

htruong, 2018-09-17:
It's time to process September payouts ($30.87). 

If you guys have no objects, all the payouts will go to the awesome @hawkeyexp.

@inspector71 I do see that you helped a lot with the wiki. If you want to get a small credit, please let us know. If you just want to help out, we would like to thank you for your help and support!

cydia2020, 2018-09-17:
Hey, I am fine without it, I literally haven't done much if anything at all, I just saw I new open issue when this was initially opened and commented in it. Thank you anyway

inspector71, 2018-09-18:
Hello

More than happy to forego any payment.

Very much appreciate the thanks for my small efforts. It is important to
read for me and is definitely helps me feel good about my contributions.

Hopefully I can contribute more in the upcoming month than the month just
gone n

hawkeyexp, 2018-09-21:
@inspector71 only short but pure truth: thx for still updating the wiki!

htruong, 2018-10-30:
Hello all,

We received a total amount of $72.13 in September from Patreon. This month is accounted for quite late, and I apologize for the lateness. As an apologize, I have decided to match the amount we received from the patrons, $72.13.

If no one objects, all the proceedings will go to the awesome @hawkeyexp.

Cheers!

inspector71, 2018-10-30:
No problems. Thanks for your generosity.

htruong, 2018-11-09:
This month's payout has come. I received 84.99 this month. If no one objects, everything the proceedings will go to @hawkeyexp :)

htruong, 2018-12-08:
This month's payout has come. I received $86.80 this month. If no one objects, everything the proceedings will go to @hawkeyexp :)

htruong, 2019-08-25:
Hello, all previous months (since Dec 7 2018), since I have done not exactly much, all the proceedings were transferred to @hawkeyexp . I have received this month Patreon payout in August for $49.99 and I think no one else deserves it more than @abraha2d.

parkerlreed, 2019-09-03:
I updated the AUR PKGBUILDs to get them in a working state

aasdk-git using https://github.com/opencardev/aasdk/tree/development

openauto-git using https://github.com/abraha2d/openauto/tree/development (Had to use that fork as the current opencardev openauto crankshaft-ng branch has hardcoded config paths for Crankshaft specifically)

https://aur.archlinux.org/packages/aasdk-git/

https://aur.archlinux.org/packages/openauto-git/ 

The month of September/2019
--
Email: parker.l.reed@gmail.com
What you worked on: AUR cleanup
How much time: 1 hour
Purchases: $0

closed 2021-11-27 by github-actions[bot]