opencardev/crankshaft #165: [csng] Custom screen resolution

rubenverhoef, 2018-07-12:
I have a 7" waveshare type C 1024 x 600 screen. It works but the crankshaft homescreen is not full size and when i press out of the picture at the black parts i get a black cross.

Also in android auto the touches in the 800x480 area are working but the rest is not responding. The android auto display is displaying over the whole screen.

Maju3, 2018-07-12:
Edit config.txt file.
On windows use notepad++

Add this to config.txt

hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 3 0 0 0
hdmi_force_hotplug=1

rubenverhoef, 2018-07-13:
Sorry I forgot to mention, I have set the config file with this:
`
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
`

Almost everything is full screen but not the home screen. The clock is not at the upper right and the default wallpaper is not full screen.
Touches out of the wallpaper "box" will result in a black cross.
Android auto interface is full screen but touches outside the 800x480 "box" don't do anything to the Android auto interface. Inside the 800x480 it works correctly.

rubenverhoef, 2018-07-13:
I'm seeing some hard coded 800x480 screen resolution in the openauto code. I think we need to adjust these values acording to the screen resolution. I'm also seeing hard coded button positions (x,y) so we need to make that also depending on the screen resolution.

hawkeyexp, 2018-07-13:
we will later have a look on it

hawkeyexp, 2018-08-21:
New gui is work in progress which also addresses scaling gui.

hawkeyexp, 2018-08-29:
Feature ready for next release.

closed 2018-08-29 by hawkeyexp

sastreae, 2023-07-04:
hello. Do you have images of the project with widescreen? how energizing that screen? so far i have the official screen of element14