opencardev/crankshaft #166: [csng] No sound (Audioinjector stereo)

rubenverhoef, 2018-07-13:
I have set the audio output with the `crankshaft` command to audioinjector zero (no stereo available but is the same), it will add the line `dtoverlay=audioinjector-wm8731-audio` to the `/boot/config.txt` which is correct.
But it does not set the `alsa state` like this: `alsactl --file asound.state.MIC.thru.test restore` or `asound.state.RCA.thru.test` (MIC for onboard microphone and RCA for line input).
More information: [Audio-Injector/stereo-and-zero](https://github.com/Audio-Injector/stereo-and-zero/tree/master/audio.injector.scripts-0.1)

When i manualy restore the alsactl it works correctly but after a reboot i have to restore it again, also after a `sudo alsactl store` which stores the current state to `/var/lib/alsa/asound.state`. But after a reboot the file is gone and the state is set to default which means no audio out/input.

I think we have to restore the correct alsactl config file if the audioinjector is used during bootup procedure. If we get the store/restore at powerup right we do not have to do this in the crankshaft application and the system will do it. But there is some kind of bug in the system which will not load the previous config.

hawkeyexp, 2018-07-13:
The audioinjector support is not fully integrated currently - will be updated next time.
Don't deal with asound.state etc. - it's useless cause in normal mode system is read only so alsa can't save states etc. - it will be included in another way.

hawkeyexp, 2018-07-14:
I have updated crankshaft management tool to `0.9.3` to pepare support.
Can you check it please?

Update to latest:
`crankshaft update csmt`
Re-setup your dac:
`crankshaft audio audioinjector stereo`

`/boot/crankshaft/alsactl.state` should now be created.

Change your startup.sh to:
`alsactl --file /boot/crankshaft/alsactl.state restore`

Let me know if all works fine to be include to upcomming builds.
Thx

rubenverhoef, 2018-07-14:
I'm not able to test it this week, next week i will do it.
I have reviewed your code and a few things:

The line [echo "state.audioinjectorpi {" > $TEMPSTATE](https://github.com/opencardev/prebuilts/blob/40abb387bd0984a50ea8ca1458eb8e769ffa776e/csmt/crankshaft#L81) shouldn't it be [echo "state.audioinjectorpi {" >> $TEMPSTATE](https://github.com/opencardev/prebuilts/blob/40abb387bd0984a50ea8ca1458eb8e769ffa776e/csmt/crankshaft#L81) with two `>`?

Second thing is, now it is set to use the Microphone input, when a user want to use the Line in Input they have to use an other `astate config`. This one should be used for the line input: [asound.state.RCA.thru.test](https://raw.githubusercontent.com/Audio-Injector/stereo-and-zero/master/audio.injector.scripts-0.1/asound.state.RCA.thru.test). Maybe it is nice to have a second option when setting up the audioinjector to specifiy MIC or RCA input.

Third thing, why are you separating the zero and stereo? They have the same config and the same hardware. The only thing i can imagine is that it is less user friendly to use `crankshaft audio stereo/zero`  or something.

hawkeyexp, 2018-07-16:
The > is correct - have a look on google for bash programming.
Easy answer: > push content to file by clearing all other | >> append content to file.
About Line/Mic input - we are talking about a mic so mic input is the right one.
Why zero/stereo - cause users often don't read doc's so it's more failsave :-)

rubenverhoef, 2018-07-16:
That's true,  >> is append and the first line have to >.
The only thing I can imagine is that someone (maybe I) want to use the line in as audio input for a preamplified microphone or something. I want to use my existing microphone from my car, I think it is preamplified so I need to use the line input. I can change the asoundstate which is located at the boot partition so it is okay but including it in the `crankshaft` command makes the support complete. ( I can also make a PR for this? If you are okay with it?)

robert5974, 2020-02-15:
Just pointing out one issue with this setup, within the /boot/config.txt the i2c parameter is incorrect.  The line should be "dtparam" and not "dtoverlay" for "i2c_arm=on".  Other than that, thanks for putting in the setup for this sound card!

commit 624e83c1 (2018-07-14): [script] Fix #166