opencardev/crankshaft #167: USB Keyboard not working?

andreacampanella, 2018-07-16:
trying the latest NG with a one plus 3 on a raspberry pi 2+

If I connect the keyboard , it works in the configuration screen but no into the the OpenAuto screen.
[As from documentation ](https://github.com/opencardev/openauto/blob/4fa68d1d7c977cccc2c4586b39f893cf9994113a/src/autoapp/UI/SettingsWindow.cpp#L237) some keys should work but apparenly are not ?

smartroad, 2018-07-19:
Possibly a stupid suggestion, but have you enabled the keyboard in the settings?

andreacampanella, 2018-07-19:
Possibly a stupid answer: which settings ? :P

smartroad, 2018-07-19:
There is a tab in the settings which (if memory serves) there is a check box in the top left to enable global keyboard commands and then you need to check which ones you want to have enabled individually.

andreacampanella, 2018-07-28:
Not in the NG , anyway , the keyboard works in Crankshaft but not in AA

hawkeyexp, 2018-08-03:
I checked it - usb keyboard is working - no config needed - for example play pause.
If something won't work with usb keyboard it must be adressed on openauto by f1x.
CS has no changes in this code part.

https://github.com/f1xpl/openauto/wiki/Keyboard-button-bindings

closed 2018-08-03 by hawkeyexp

Bkmz2017, 2019-07-23:
Help! I have same problem, keyboard doesn`t work in OpenAuto screen, but when I reboot it, I see in console my presing. And keyboard work in configuration screen and in console.
Actually, it worked when I just installed Crancshaft and worked perfectly, then something happened and it doesn`t work. I have changed raspberry, keyboard, sd-card and update Crancshaft image. It still no working. I have no idea why. 
And of course I have enabled the keyboard in the settings.

Miziol, 2023-10-02:
I have same or similar problem. Everything is ok on autoApp screen, but on AA screen only 1 and 2 button work (scroll wheel), but other buttons have no reaction. In settings every thing is ok