opencardev/crankshaft #174: [CSNG] Shutdown_Button service hangs system on boot

Fotchman, 2018-07-21:
Not sure if this is a CSNG issue or something else.
On a clean install of 2018-07-13-crankshaft-ng-9B0FBA9 after installing the "reboot/shutdown script from github https://github.com/scruss/shutdown_button, the system hangs on the splash screen during boot. Most times a power off and back on will let the system boot through (but not always), and randomly the system will hang at the CLI prompt. Buttons never do work, sometimes also locking the system. Disabling the service and rebooting/halting from CLI works every time. Tried an alternate GPIO with the same results. I'm no expert, but I was able to recreate this multiple times on multiple clean installs. All tests while in Dev mode.

Installed hardware:
PI3
Galaxy S9+ 
HiFiBerry
Official 7" screen

Maju3, 2018-07-21:
Have you tested the script with crankshaft versions before crankshaftNG ?

Fotchman, 2018-07-21:
Maju3 - just booted the latest release of master CS and the shutdown and reboot timers work each time. No hangs or other issues. Will reload previous versions of CSNG to see if/when the feature breaks.

hawkeyexp, 2018-08-03:
The pre4 this weekend is on its way and includes a gpio based shutdown option with cunfigurable delay time. Try out if it is usefull for you and if solved close the issue please.

Fotchman, 2018-08-03:
Thank - will check it out and let you know.
Thanks for everything,

closed 2018-08-07 by Fotchman

hawkeyexp, 2018-08-07:
You're welcome