opencardev/crankshaft #178: [CSNG] Custom brightness control script

rubenverhoef, 2018-07-23:
It would be nice if a user made brightness control script/command is launched every time the brightness changed with the brightness value as an argument.

So i can make a script with `gpio pwm 26 "brightness value"`

see #176

hawkeyexp, 2018-07-24:
Will be available with pre4.

hawkeyexp, 2018-08-03:
Pre4 is available this weekend including the feature.

osokthedevil, 2018-08-06:
for some reason i can't even adjust the brightness with the latest update? using this display that worked in older versions that i can adjust the display brightness. https://www.amazon.ca/gp/product/B014WKCFR4/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

hawkeyexp, 2018-08-06:
@osokthedevil enable debug mode in settings, reboot, adjust brightness, create debug log and add here please.

osokthedevil, 2018-08-06:
sorry @hawkeyexp dumb question but how do i extract the log file?

osokthedevil, 2018-08-06:
[debug.zip](https://github.com/opencardev/crankshaft/files/2263278/debug.zip)

hawkeyexp, 2018-08-06:
you can grab it from sd /boot/debug.zip or connect via ssh to copy it somewhere

osokthedevil, 2018-08-06:
sorry haven't finished my first cup off coffee. debug.zip file dropped for you

osokthedevil, 2018-08-06:
ya doesn't move at all. everything else responds to touch but the slider for some reason.

hawkeyexp, 2018-08-06:
Same on day and night ? Try to switch to night (auto switching back is only every minute) and test it there please - currently i have not really an idea...

osokthedevil, 2018-08-06:
yes day or night mode no change, ya a little weird. I was using crankshaft -2018-03-13 before i updated yesterday. i know that one works fine with the display. Hopefully that helps you out some what.

hawkeyexp, 2018-08-06:
Please let me know if your system creates the file /tmp/custombrightness

osokthedevil, 2018-08-06:
no file found. It also says in the settings under - custom brightness command = disabled

hawkeyexp, 2018-08-06:
Please check if a direct control is working via shell as pi user:

this should set max brightness:
`echo 255 > /sys/class/backlight/rpi_backlight/brightness`
this should set dark brightness:
`echo 55 > /sys/class/backlight/rpi_backlight/brightness`

osokthedevil, 2018-08-06:
will do so. I just tested the release before the new one, 2018-07-13-crankshaft-ng-9B0FBA9.zip and the brightness adjustment works. going to reload the new release and check those settings

osokthedevil, 2018-08-06:
okay so running the echo commands works for day and night adjustment. At least give me some what of the ability to adjust the brightness at the moment.

osokthedevil, 2018-08-06:
[debug.zip](https://github.com/opencardev/crankshaft/files/2263502/debug.zip)

here is a new debug.zip file for your viewing

hawkeyexp, 2018-08-06:
There seems to be no prob i can identify currently.

osokthedevil, 2018-08-06:
okay, i don't know what else i can say beside 07 13 version brightness slider works fine and 08 03 it doesn't.

hawkeyexp, 2018-08-06:
I'm sorry at the moment - it is the first feedback about a brightness prob currently - i keep an eye on it.

hawkeyexp, 2018-08-06:
I have update the management tool to collect some more info.
You can update it with `crankshaft update csmt` in shell (you need internet access)
After update (1.1.4) you can create a new debug log. Possible it gives a better result.

osokthedevil, 2018-08-06:
Crankshaft Management Tool v1.0.9 is executing command...

New file:
Current file: 8f402dcabf5e78b6829e8db2c8e9926f
ABORT: checksum fail.
pi@crankshaft:~ $

hawkeyexp, 2018-08-06:
Try later sometimes github needs a bit to resync - but a note - debug mode has no dns access - do it in dev mode.

lockeandev, 2018-08-07:
Same problem confirmed, brightness slider worked on Pre3 but not responsive in Pre4

hawkeyexp, 2018-08-07:
Here is a preview of next build including a slider fix which should work - please try out and report back. Thx.

[https://drive.google.com/file/d/1ZvIEtntpNv3t4zlDqPSyb_2HypJH1Ck4/view?usp=sharing](url)

hawkeyexp, 2018-08-08:
@osokthedevil does the preview fix your problem?

osokthedevil, 2018-08-08:
Sorry I haven't tried. Plus my knowledge is limited to even the debug and using that as well. I'm finished Holiday so been busy at work

osokthedevil, 2018-08-09:
@hawkeyexp downloading now, i will let you know in a bit.

osokthedevil, 2018-08-09:
@hawkeyexp brightness is working properly now

osokthedevil, 2018-08-09:
@hawkeyexp love the wifi add ins

hawkeyexp, 2018-08-14:
slider is fixed custom brightness, too -> closing.

closed 2018-08-14 by hawkeyexp