opencardev/crankshaft #179: [CSNG] GPIO shutdown timer

rubenverhoef, 2018-07-23:
It would be nice if we have a GPIO shutdown button with a delay time. So when i turn my car off and the GPIO input is going to 0 (or 1 option for high or low trigger would be nice also) the RPI shuts down after a given time delay.

So it is just the same as the phone disconnect timeout shutdown but now with GPIO input as toggle.

rubenverhoef, 2018-07-25:
No it is not the same idea. When i have an iginion detect input and it is more than 5 sec off it will do nothing unless it is on again. I´m seeing more possibility for intergrading it into Crankshaft NG.

I can make a PR for this with the use of a separate crankshaft timer, so the same functionality as the phone disconnect timeout put with a gpio status input.

I can also make my own scripty but i think most people need this. I want to turn off the system when i shut down my car after a few minutes. And i don´t want to have the system turn itself off when i disconnect my phone because i´m not always driving with the phone connected and i want to use my rearcam.

hawkeyexp, 2018-07-25:
I know what you mean ge sure. I Use the other issue only as reminder and will start possible today or tomorrow to release a solution.

hawkeyexp, 2018-07-25:
contact me @hangouts hawkeyexp

rubenverhoef, 2018-07-25:
google hangouts? how can i search for you?

hawkeyexp, 2018-07-25:
-> new conversation - type in name

rubenverhoef, 2018-07-25:
No results, but your email works i think