opencardev/crankshaft #181: several issues noticed during long drive (version July 1st)

partach, 2018-07-25:
- Phone: Samsung S9+ with latest android. July 1st version crankshaft
- rpi 3b+

1) Navigation assist voice out of sync. (not sure if this is a crankshaft thing) This happens most of the time. It looks like the entire audio stream for the route voice over is pre-loaded and a certain time slot is played of it each time. What happens is that it gets out of sync so you hear a last part of a previous direction and only a part of the current. Very annoying. The next direction continues where the previous had stopped.
2) Screen watchdog when connected. When connecting the phone most of the time the watchdog for switching off the screen still works. very annoying to need to keep pressing the screen each 5 minutes on a 12 hour drive when needing to navigate. This happens most of the time that the screen turn black after about 5 mins. Default the screen should stay on when phone is connected via USB.
3) when selecting another screen (music, tel, etc.) the screen hangs while building up. About 80% of the sreen has roled up and is still a bit transparrent and then hangs. Unit is then un-responsive for a while. This happens sometimes.
4) spontaneous reset. It happened twice in 2 weeks (total distance travel 3500km with about spending 30 hours in the car in total. Out of the blue the system did a reset just while i needed to navigate through Paris. This was not an ideal moment.
5) Ók google hangs. When asking google for help sometimes it just hangs and only shows the four dots. You need to tap the screen for it to go away. Asking. 'ok google' a few times in a row makes it unresponsive as well.
6) Unable to voice messages. When pressing to play a received whatsapp it stays quit. The 'ok goolge' screen shows with the dots but nothing happens. You need to press the screen to get rid of it.

Hope this helps, let me know if more information is required.
Regards

tehniemer, 2018-07-25:
I've expereienced 1, 5 and 6 also

hawkeyexp, 2018-07-25:
Thx for feedback. But many probs are fixed since build from 1st july (Pre1) - currently we are on pre3 and pre4 will be available on the weekend. Please upgrade to latest build to be sure we are not hunting old fixed probs. Thx again - i will close the report cause it is based on outdated version.

closed 2018-07-25 by hawkeyexp

Maju3, 2018-07-28:
I have noticed 1,5 and 6 on csng pre3.
So i wouldn't say this report is based on outdated versions.