opencardev/crankshaft #182: [csng] [Feature request] Day / Night button to change screen brightness along with wallpaper

pumpkintechlabs, 2018-07-28:
I find it quite often needing to lower the brightness of the screen at night and it is a bit awkward to do with the current implementation using a slider and requires taking the eyes off the road for longer. Would be good to be able to change day / night with one button click.

The button that currently changes the wallpaper from day to night could change the screen brightness as well to a predefened value in Settings. (Two brightness settings for day and night)

Thanks and keep up the great work!

hawkeyexp, 2018-07-28:
This feature is included. The levels are saved if you hit the button to toggle day/night.

Be in day mode - set brightness level via slider.
Switch to night mode (day level will be saved now)
Now you are on night mode - set brightness level via slider.
Switch to day (night level will be saved now and last day level restored)

Generally last value before switching is saved every time you switch.

hawkeyexp, 2018-07-28:
@smartroad - No it's not currently.

closed 2018-07-29 by pumpkintechlabs