opencardev/crankshaft #183: Audio settings are not saved

andreacampanella, 2018-07-28:
I "un-tick' the Audio settings , and then I press save , it return to the main screen , I connect my phone , everything is fine , the audio is routed (as , not routed ) as I want , but if I restart the raspberry pi , the settings are lost , everything single time I have to in select the audio settings and then save , it never saves them.

hawkeyexp, 2018-07-28:
You have to add info's about version etc - in this style nobody can hell you!

cydia2020, 2018-07-31:
Same here, alsamixer settings were not saved, real pain in the back.

andreacampanella, 2018-08-01:
Yes , sorry , I'm using crankshaft-ng pre 4

hawkeyexp, 2018-08-01:
No pre4 is out - it's work in progress so opening issues makes no sense - sorry.
Issues must be based on published releases.

closed 2018-08-01 by hawkeyexp

andreacampanella, 2018-08-07:
So, you are not collecting bugs for your dev release? how you are going to find bugs? guessing?

hawkeyexp, 2018-08-07:
Not in this way...
- no information about something in initial post
- talking about a pre4 which was never released and is cunfusing other users