opencardev/crankshaft #185: [CSNG] Feature Request:  On display volume control

chrisfromwa, 2018-08-02:
Originally I was hoping to use some type of physical volume control for this project.
But the more I thought about it,  the more I think having it all on the display would look nicer in the car.  

Would it be possible in a future release,  to include a on-display volume control? 
Maybe something that could be turned on/off in the setting for those who don't want to use it?

I created a crude mock-up of what i was thinking,  this would be very useful for those who are trying to do a full head unit replacement.  I'm sure it could be made to look nicer and OEM almost.  THis was just a rough idea.   Basically I just have a sidebar,  where if you press the speaker +  the volume goes up,  if you press the volume -  it goes down.   Maybe also a 'mute'  button would be useful too. 

[image]

hawkeyexp, 2018-08-02:
Would be nice but with my current skill i can't release it.

chrisfromwa, 2018-08-03:
Are you saying there is no way this could ever become a feature?
Is it because you're not sure how to implement something like this, or is it physically impossible to do?  

Are there any other developers on here that would be willing to look into a solution?

glennsaint, 2018-08-03:
@chrisfromwa I think you should request that to the developer of openauto.

chrisfromwa, 2018-08-03:
@glennsaint   
Do you know where feature requests for OpenAuto are made?  
I've seen their thread on XDA,  but not an official issue/request area like this. 

This seems like a very much needed feature,  as I've seen many who would like to use this as a headunit replacement, and a way to control the volume via the software would really take this closer to being able to do that.

hawkeyexp, 2018-08-03:
https://github.com/f1xpl/openauto/issues

I have opened an issue as question.

chrisfromwa, 2018-08-04:
@hawkeyexp 
Thank you for posting on the OpenAuto forum,  from the response by F1 it looks like it may be possible?
Do you think this is something that could be looked into further for a future release?

I want to do a CrankShaft install in my older Camaro,  and I have limited space.  Physical buttons, or knobs I think just won't look right and make it harder to install.   If there were onscreen options I could start modifying the cars bezel and get the sreen installed pretty quickly and make it look like it was OEM. 

Thanks in advance.

hawkeyexp, 2018-08-04:
If possible and my skill is enough it will be added ;-)

chrisfromwa, 2018-08-06:
I have faith in you!  :)

hawkeyexp, 2018-09-27:
I will close for now cause still not possible but on the list and if a way find it will be relased.

closed 2018-09-27 by hawkeyexp