opencardev/crankshaft #186: [CSNG] Customizing Buttons

glennsaint, 2018-08-03:
Hi, is it possible to edit/modify the appearance of the CS Buttons? I am using pre3. thanks!

hawkeyexp, 2018-08-03:
On a later time i will have a deeper look on it what is possible - ideas are present but not ready.

petterhj, 2018-08-06:
Does QT somehow allow loading the UI-files (e.g. autoapp/UI/mainwindow.ui) and/or resource files at boot time? This could possibly at least allow users to add (graphics, static text) and reposition/hide existing elements as well as "skin" the user interface in general.

hawkeyexp, 2018-08-06:
It seems there is a class for this case of use: http://doc.qt.io/qt-5/quiloader.html but nothing included currently.

hawkeyexp, 2018-08-21:
New gui is work in progress which also addresses scaling gui and custom buttons.

hawkeyexp, 2018-08-29:
Feature ready for next release.

closed 2018-08-29 by hawkeyexp