opencardev/crankshaft #187: WaveShare screen, touch rotated 90 degree

Theklinto, 2018-08-03:
So, I have a plan to build a car PC for my Peugeot 406, but I ran into a problem when i wanted to setup my Raspberry Pi 3+.

I installed the touch drivers from WaveShares website, I'm using their 7 Inch 800x480, but when i tried to use the touch, it was rotated 90 degree to the right compared to the screen.

I've uploaded a picture to imgur trying to explain what is happening.
https://imgur.com/a/Y5m0GXG

When i'm touch the red area, it actully registers the touch as in the purple area.
The screen I'm using is using the pins on the Pi and not usb.

I'm totally new to Linux and Raspberry, so bear with me.
Here is what i tried:
use the display_rotate=1/2/3
 the setting in the crankshaft config, option to rotate.
I've tried to rotate the screen through manufactor drivers, and i can rotate it to portrait where the touch is accurate, but i want it in landscape mode.
 
-Klinto

CalmBSK, 2018-08-03:
display_rotate= only rotates the display but not the touch input.
lcd_rotate= rotates both
however, for Crankshaft I would take a look at the following in the wiki -
https://github.com/opencardev/crankshaft/wiki/Custom-touchscreens

hawkeyexp, 2018-08-03:
Possible you have to switch x/x touch input axis.
If your touch is new and you can send it back do it - for beginners not optimal.
I recommend the orginal touch from rpi foundation or a touch which is fully hotplug and driverless - hdmi + usb hid input. For Example i have a waveshare display, too - but the 7" 800x480 LCD(B) Rev 2.1 - fully driverless and working without any problem.
If a display needs special drivers etc. it is a hint that you will run into probs if you have not the skill to configure linux and it's hardware.

Theklinto, 2018-08-03:
CalmBSK:
when ever I try to start with in X11 mode it crashes and can't start crankshaftOS

Hawkeyexp:
Unfortunately I can't send it back, since it would cost as much as the display. I tried being cheap buying a screen from China. Since I'm new I though it had no impact.
Is it possible to do myself? Spent 50$ on the screen and don't want to spend another 100$ for another screen.

CalmBSK, 2018-08-03:
Sorry Klinto, I should have made this clear in my original post.
I am in no way an expert but was aware that, at least for the official RPI screen, that lcd_rotate needed to be used not display_rotate.
I then added the Crankshaft wiki entry as Crankshaft appears to actually use FLIP_SCREEN= within its config file rather than either lcd_rotate or display_rotate.

Hawkeyexp, rather than me, is the person you want to listen to so maybe they can add something regarding X11 mode but I would assume they are very busy and trying to talk someone through something like this could be very time consuming and very difficult without having access to the screen themselves. 

It may be worth contacting WaveShare to ask them about the ability to switch the touch input axis as per Hawkeyexp's idea.

If you are lucky maybe someone else will chime in who has resolved this problem themselves or at least knows how to.

Good luck!

CalmBSK, 2018-08-03:
One last thing to add -

I've just checked the following -
https://www.waveshare.com/wiki/7inch_HDMI_LCD_(B)#Rotation

There is a section regarding rotating the screen and then a section about rotating the touch input. 
Have you tried that?

Theklinto, 2018-08-04:
CalmBSK
I've actually tried that. But whenever I reach step 3 it says the destination doesn't exist since I don't have a usr folder.

CalmBSK, 2018-08-04:
That's strange as I attempted to follow the steps up to but not including step 5 and was able to do so.
Step 1 and 2 were of course not required as xserver-xorg-input-libinput was already installed.

I'm assuming you have enabled dev mode in chrankshaft_env.sh otherwise you would not have been able to ssh into it.

Not sure what to suggest other than to try once from from a fresh install.

Theklinto, 2018-08-04:
CalmBSK:
Yes I've entered Dev mode and using PuTTY as ssh client.
I'll try to do a fresh install and try your suggestion. But is there anyway to start crankshaftOS from ssh? I'm currently resitting the SD card every time to exit Dev mode

hawkeyexp, 2018-08-04:
There is no need to resetting your sd every time - just reboot - i think you should move to reddit for further discussions how to get it working.

closed 2018-08-04 by hawkeyexp