opencardev/crankshaft #192: Google voice out of sync

partach, 2018-08-08:
Phone S9+ latest android (8.0), rpi3b+, CS pre4

When navigating a route via google maps the (dutch) direction voice always gets out of sync at some point.
When this happens It starts by playing the last part of the previous direction and only tells a part of the new direction and so on. This makes the voice direction useless and very distracting.
Google assitant (via 'ok google') can have a similar effect. When asking for something it stays quite (animated color buttons are visible). When then again asking for something a bit later it gives the feedback of the previous question.  

When it is out of sync it seems that whatsapp playback does not work anymore (no voice heard) and needs to be dismissed by pressing somewhere on the screen as the screen stays on the color buttons otherwise.

This i observered in version pre1 (reported there too but got dismissed as it was not the latest anymore), pre3 and now pre4 (pre2 not tested).

tehniemer, 2018-08-08:
Try switching the audio settings from RT Audio to Qt or vice versa and report back. Other users have reported success with this.

hawkeyexp, 2018-08-09:
Switching audio to try out is the only check you can do. If it won't fix your prob cs can't do anything for you. But you can open an issue on the openauto repo from f1x - it's the core used in cs.

Maju3, 2018-08-09:
In my case switching to RT audio has solved this "sync" issue.

internix14, 2018-08-09:
Nothing helps in my case, not even switching from RT to QT vice versa.

unxpctederr, 2018-08-11:
(Pre3) I have this issue too. It only happens on google maps and not waze. It happens when I perform a voice command using ok google/touching the mic icon, the maps voice navigations will become out of sync. If I do not use voice commands it stays in sync. I will test it on Pre4 and also switching audio to Qt.

partach, 2018-08-18:
I have now tested both RT and QT.
With RT it seems worse (occurs more often/quicker) but i have the problem with both.
Also Spotify is affected. Sometimes when going the next song after one is finished it remains quite. The display does indicate the next song is selected, but no audio. If you press for a next song at that point, when you have figured out it is quite for a while, it displays the next song but plays the previous (the one that did not seem to start). From then it is out of sync. (Showing the next song as text and display but audio belongs to the previous song in the playlist.)
So it seems to be with all audio that 'a sync issue' will occur at some point. 
I play audio via a USB dongle. Could that have any influence?

hawkeyexp, 2018-08-18:
No idea sorry - Today i have done some tests during a drive with a friend.
Setup: RT-Audio / Hifiberry DAC Plus / Playing Music via Google Music/TuneIn Radio / Navigation active with GMaps
Result: Playing Music was without problems - auto playing next tracks works fine. During voice commands from maps music was leveled down - > voice command -> level up again for music.
Voice commands always on correct position during drive.
Only some init scratch on first milliseconds of voice command that's all but only on init not during whole command.
Only one chance - try another audio hardware.

I will close for now cause it seems not a general prob.

closed 2018-08-18 by hawkeyexp