opencardev/crankshaft #193: Timezone not saved / day-night mode RTC relation

partach, 2018-08-08:
Phone S9+ latest android (8.0), rpi3b+, CS pre4

In the pre4 settings menu you can select another timezone.
However this never gets saved (pressing save does not matter, it is always reset to default after reboot).

Can it have a relation with an absent RTC?
On that subject: Would it be possible to look at the time of the phone in relation with day/night mode settings. It seems that without RTC day/night mode does not do anything. It would be great if the day/night mode settings are used based on (follow) the time of the connected phone.

hawkeyexp, 2018-08-08:
- Timezone is used to make an rtc working correctly. Without an rtc it is useless.
- You can enable developer menu in android auto running directly on your phone and enable device controlled day/night.

closed 2018-08-08 by hawkeyexp

Grawa, 2018-08-24:
hawkeyexp Great solution, but (this is a **"feature-request"**) 
can you simply add a setting in "settings tab" to **disable clock widget** on desktop?

hawkeyexp, 2018-08-24:
@Grawa - have you ever checked the general tab :-) ?

Grawa, 2018-08-24:
checked now, lol,sorry🤣 thank you 😆