opencardev/crankshaft #194: [csng][Feature Request] RW exFAT and or NTFS support

pumpkintechlabs, 2018-08-08:
Trying to upgrade from pre3 to pre4 with a 64GB USB flash formatted in exFAT doesn't work as the drive is not being recognized. Haven't tried NTFS yet. Is there any other file system formats i should try?

Would be good to have support for those file systems as well as for CSSTORAGE formatted with those file systems to run Kodi off.

hawkeyexp, 2018-08-08:
Some days ago the wiki was starting to be updated...
[https://github.com/opencardev/crankshaft/wiki/Installing---Updating](url)
The current limits to ext3/ext4/fat32 is depending on the buildin drivers from kernel. exFat and NTFS are no buildin drivers. During flash the system is starting with a minimal system in a ramdisk without
the drivers and such fs wouldn't be accessible. In a later step i will extend the file systems which are supported - but only for CSSTORAGE not for flash.

hawkeyexp, 2018-08-09:
The package is not enough. The Service which handles mount of CSSTORAGE (including fsck etc.) must also be reworked.

hawkeyexp, 2018-08-09:
I have done some checks but have to revert the announce - exfat has no way to repair in linux so it is useless cause if filesystem is demaged it won't be mountable anymore or cause other probs until it is repaired on another system. I will skip both cause it's not native supported.
Use ext3 or ext4 for bigger drives. Sorry.

closed 2018-08-09 by hawkeyexp