opencardev/crankshaft #199: Crankshaft not booting into Autoapp after DevMode

SirWalllace2, 2018-08-11:
I tried to install a driver for my touchscreen [this one](https://www.ebay.de/itm/1024x600-7-inch-HDMI-Display-Raspberry-Pi-LCD-IPS-Capacitive-Touchscreen-win-en/283043059090?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) because the touch functions doesn't work. But after installing the drivers and switching DEV_MODE to 0 aigan my installation won't start to the Autoapp anymore. It just wait for me typing Username.

When I tried it first, everything worked fine except that the touchscreen did not respond. Switching to X11 didn't worked either. So i tried it with those drivers. But now i can't start the app anymore.

Google Pixel 2 XL
Android 9
Raspberry 3 Model B
Crankshaft version 2018-08-03

hawkeyexp, 2018-08-11:
Sorry but you killed your installation cause this "installers" overrides files without checks which works ONLY on a fully unmodified raspbian and ONLY on the version they tested it.
With this type of display you doesn't need any driver cause it's HID complient. I have also one of this type from waveshare which works out of the box. You ony need to configure the display values in config.txt thats all.
As a sidenote: this types of display are really power hungry and also bad cables kills touch input.

closed 2018-08-11 by hawkeyexp

SirWalllace2, 2018-08-13:
Ok,
i've got the mistake. It really was the power source. I used my powerbank for second power in addition to the Raspberry power source over USB, but it deactivated the Touchscreen. With a wall plug it works just fine.... I would have never thought of that. Thank you

hawkeyexp, 2018-08-13:
No prob :-)