opencardev/crankshaft #202: [CSNG][Feature Request] Option for verbose boot

pumpkintechlabs, 2018-08-15:
On previous iteration of crankshaft (pre csng) I was able to have verbose boot at startup. Would be good to have an option that shows verbose boot without any splash screens if possible. The option can be set at boot.txt and will disable the splash screen and show verbose boot.

Cheers!

hawkeyexp, 2018-08-15:
Hm.. what should be the benefit ? You can access the logs and will see what happens during boot.
Also you can disable the splash yourselve.

pumpkintechlabs, 2018-08-15:
Saves having to access the logs especially if it's already installed in the car. On the other hand some people like myself like the verbose boot instead of splash screens - i disable them all the time when i run linux.

hawkeyexp, 2018-09-21:
i think won't be included. If you need the output try removing the "splash" in cmdline.txt

closed 2018-09-21 by hawkeyexp