opencardev/crankshaft #203: PiCam is not working correctly

LoganFanGer, 2018-08-18:
First of all thanks for the new release. That looks very good.
Hifiberry support works, RTC works. Volume and Next / Previous track via GPIO-buttons works great.

The problem I still have is the RearCam. I have connected to GPIO19. The button is connected to GND. When I start CS immediately comes the camera image, goes out, goes on .... changes about every 2 seconds. If I press the button, the cam remains on! Does anyone have an idea what's going on?

With my own script, which I have included manually, the cam works without problems, but without the great new Dashcam function.

Greetings from Germany, thank you for your good job.

hawkeyexp, 2018-08-18:
Do a check with crankshaft command via shell.
Disable cam in hw setup, enable debug mode and reboot.
Connect via ssh to shell and enter `crankshaft gpio all`
This way you check the live state of the gpio's.
Sound's like anything else is using your gpio and conflicts.
Make also sure the button is open in normal state and to trigger the cam it is closed and connects to ground (seems to be your setup).
Greetings from Germany ;-)

LoganFanGer, 2018-08-18:
[image]

Thank you for your prompt reply.
That's really strange indeed! Connected are GPIO5 and 6 for forward / back and 12 and 13 for the volume. The switching states are also displayed correctly. GPIO19 reacts correctly to the press, sometimes not. That would explain the flickering of the screen! 

What I do not understand is that other GPIOs are also switched, although there is nothing connected. Incidentally, the status of GPIO18 is sometimes on or off, without a button being connected.
I'll try another PI, maybe my one has a defect! I think it's my hardware!

Du kommst auch aus Deutschland?

hawkeyexp, 2018-08-18:
It's no defect - some gpio's have a default state 1 some 0 that's normal on a pi.

> GPIO19 reacts correctly to the press, sometimes not.

Note: The gpio can be used by something other - have a look here:
[GPIO19](https://de.pinout.xyz/pinout/pin35_gpio19#)

So test another gpio and possible another switch.

Seems the prob is not caused by system so closing.

Ja :-)

closed 2018-08-18 by hawkeyexp