opencardev/crankshaft #204: [CSNG][Issue]Doesn't save changes in CS HW tab

bttf92, 2018-08-19:
Just flashed the new Pre6 image (clean flash, not updated) and the settings in CS HW tab are not being saved. From my tests seems that all the settings from the other tabs are saved successfully.

I have attached the debug.zip made after save
[debug.zip](https://github.com/opencardev/crankshaft/files/2300422/debug.zip)

hawkeyexp, 2018-08-19:
Have you checked the box on the right side ?

bttf92, 2018-08-19:
Tried now. With the "exec setup on save" checked only timezone isn't being saved (tried again also with exec setup off), other settings are now being saved (also bt works).

[debug.zip](https://github.com/opencardev/crankshaft/files/2300521/debug.zip)

hawkeyexp, 2018-08-19:
I have tested it and it works.
Can you try it in a ssh shell ?
Boot into debug mode and connect via ssh:
enter `crankshaft rtc ds1307`
let me know if you get any fail - you can check the timezone in /etc/timezone

bttf92, 2018-08-20:
crankshaft rtc ds1307 returns this error:

```
Crankshaft Management Tool v1.3.0 is executing command...

/var/lib/dpkg/info/tzdata.config: 337: /var/lib/dpkg/info/tzdata.config: cannot create /etc/timezone: Read-only file system
Reboot needed to activate!

Done.
```

hawkeyexp, 2018-08-20:
please do a fsck on root partition if there are errors which prevents from switching between rw and ro

bttf92, 2018-08-20:
Sure, will do when I will be at home. In the meantime, if there is a problem with the partition, why only the timezone setting fails to save?

hawkeyexp, 2018-08-20:
It was a try but currently i could reproduce a possible problem - let me check it...

hawkeyexp, 2018-08-20:
Update csmt with `crankshaft update csmt` - problem should be fixed.
Please report back and close issue if fixed.

bttf92, 2018-08-20:
Works! Thank you!

closed 2018-08-20 by bttf92