opencardev/crankshaft #205: [CSNG][Feature Request] Python Serial Library Inclusion

t00fatt, 2018-08-19:
I am using an arduino micro to interface my steering wheel buttons with my head unit and CSNG. I am communicating with the arduino over USB serial sending data to the arduino, like the phone connection status among other things. However, steering wheel control commands from the arduino are sent to the RPi as keystrokes, and not over serial.

The python serial library is not included with the CSNG build, and after doing a USB update it seems the library is gone again and must be reinstalled via dev mode. I would like to avoid having to ssh in and install the library with every update. 

Is it possible to include the python 2.7 serial library with CSNG builds or at least make it so added libraries remain intact after minor updates? Anyone using an arduino with CSNG can benefit from the inclusion of the serial library.

hawkeyexp, 2018-08-19:
I agree, the option to commincate with an arduino could be usefull for some more users.
Which deb package is needed to be included ? It is easy to add.

About port manually installed packages during update:
This could not be done, cause usually no internet connection is available during this process so an auto install will fail, additional it is hard to decide which packages are really needed.
I will think about a possible solution which could be helpfull for this case but it is sure that a manual step will be needed.

t00fatt, 2018-08-19:
I've been using pyserial

https://github.com/pyserial/pyserial/

t00fatt, 2018-08-19:
Sorry, yeah that is what I should have linked to.

hawkeyexp, 2018-08-19:
Included in next build.

closed 2018-08-19 by hawkeyexp