opencardev/crankshaft #206: [CSNG] Pre6 Write Access Disabled in Dev Mode

t00fatt, 2018-08-20:
- What crankshaft version?
NG Pre6
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible.
Clean install
Boot into dev mode
no write access
use CMT filesystem boot unlock, "done" returned
still no write access on boot

I can not enable write access on either boot or system in dev or debug mode after updating to Pre6. Worked perfectly in Pre5 with write access in dev mode on boot.

[debug.zip](https://github.com/opencardev/crankshaft/files/2300840/debug.zip)

hawkeyexp, 2018-08-20:
Tried with the published build - all ok. Log seems also clean.
I only had this problem one time a while ago which was caused by a corrupted fs on /boot.
I updated csmt and added a more clean output with state before and after remount and a check option for boot.
Update it in dev mode with:
`crankshaft update csmt`
Run:
`crankshaft filesystem boot check`

hawkeyexp, 2018-08-20:
Possible a demanged sd but also possible only it was an fs fail.