opencardev/crankshaft #207: [CSNG] Feature Request - Bluetooth output from Crankshaft to head unit

Fotchman, 2018-08-21:
Good day kind devs! Love how the project is progressing. Bluetooth streaming from Galaxy S9+ to Crankshaft/RPi works great, no issue there. I would love to be able to output via Bluetooth to my head unit (since it doesn't have an AUX in). Is this, or will this, be possible in a future build? The input streaming seems redundant since for AA to work, you have to be cabled in via USB. Not complaining in the least, just seeing is this will be possible. Aloha!

Maju3, 2018-08-21:
Already possible.
Pair your phone with your car stereo BT.
In crankshaft go to settings tab"audio".
Uncheck both channels(music,speech).

Fotchman, 2018-08-21:
Hi Maju3 - thanks for the quick response. I did the steps outlined above, but I don't see how to pair Crankshaft to my HU. As a test, I'm trying to pair Crankshaft with a JBL BT enabled speaker. The JBL will send out a 'hello' message but expects the other device to begin the paring process. I'll check my HU tonight to see how it pairs (new in box). If it can initiate the BT response then all is well.

Maju3, 2018-08-21:
No, you need to pair your PHONE to your cars bt stereo.
NOT crankshaft.

Maju3, 2018-08-21:
That means you pair your samsung galaxy with your cars HU.
NO need to pair your phone with crankshaft.

Maju3, 2018-08-21:
The bt pairing phone to crankshaft is for people who have NO bt car HU;)

Fotchman, 2018-08-21:
OK, that's the 'feature' I was asking for :)  I didn't think it was currently possible but would be awesome to transmit BT audio to the HU. One less set of cables to deal with. The phone-to-Crankshaft pairing works fantastic.
Thanks again,

Maju3, 2018-08-21:
Jup.
As said if you have a bt HU then there is no need to pair your phone with crankshaft.
Just pair your phone with your HU and you are fine.

Maju3, 2018-08-21:
And in cs audio settings just uncheck music- and speech channel

Maju3, 2018-08-21:
I think you can close this "issue" :)

hawkeyexp, 2018-08-21:
As maju3 explained do a direct pairing of you phone with your car's HU. Cable and bluetooth is not really redundant cause phone calls are only possible via a bluetooth connection.

closed 2018-08-21 by hawkeyexp