opencardev/crankshaft #208: Audio doesn't sound right since Pre4

dreamvictim, 2018-08-21:
First off I'd like to thank you for such excellent work on this project but I've noticed an audio problem since the release of Pre4.

My Audio config:
Hifiberry dac+
USB Mic
RT audio (If I use QT the audio quality is great but channels drop out and sound disappears)

Everything sounds great on Pre3 but if I use any release after Pre3 including Pre6 the audio sounds flat. It's hard to explain but if you play your favourite music (loud) on Pre3 and then compare it to Pre6 you will definitely hear a difference. Can anyone else confirm if they are hearing the same? In my testing the audio quality is so much more clearer on Pre3.

Maju3, 2018-08-22:
I just have compared pre3 with pre6 now.
I use an Allo Boss Dac.
Both versions testet with RT audio.

I can absolutelly confirm that pre3 sounds much better.
The music is clearer and sounds "richer of details".
Hard to explain but in pre3 the music is more "present" and has more "punch" while it sounds more flat in pre6.

hawkeyexp, 2018-08-22:
@all feel free to test changes inside pulseaudio `daemon.conf` and report back.

Until pre3:

`default-sample-rate = 48000`
`resample-method = soxr-hq`
`default-fragments = 8`
`default-fragment-size-msec = 5`

Pre4 and up:
`default-sample-rate = 48000`
`alternate-sample-rate = 16000`
`resample-method = soxr-hq`
`default-fragments = 8`
`default-fragment-size-msec = 5`
`flat-volumes = no` -> this only prevents for volume override >100%.

dreamvictim, 2018-08-22:
Ok, I found a post about improving pulse audio quality. [Here](https://medium.com/@gamunu/enable-high-quality-audio-on-linux-6f16f3fe7e1f)
I added the following lines to the daemon.conf below the existing crankshaft config.

default-sample-format = float32le
default-sample-rate = 48000
alternate-sample-rate = 44100
default-sample-channels = 2
default-channel-map = front-left,front-right
resample-method = soxr-vhq
enable-lfe-remixing = no
high-priority = yes
nice-level = -11
realtime-scheduling = yes
realtime-priority = 9
rlimit-rtprio = 9
daemonize = no

A quick test seems to improved the audio quality a lot. Can someone else try it and confirm?

dreamvictim, 2018-08-22:
I just went for a drive so I could crank the volume up and test different styles of music. It's definitely a huge improvement and now I think it sounds clearer than pre3. Only problem is my ears are ringing lol.

hawkeyexp, 2018-08-22:
A general note:

sure pulseaudio needs some finetuning for csng. But please make sure it is working generally for everyone so as minimum it doesn't kill onboard audio if your are using an dac ur usb dongle. Not every hardware can handle same params and please filter useless parts which has nothing to do with the soundquality like the nice level daemonize or channel mappings.
Most helpfull is posting only params which are really changing soundquality than it is a good base to merge them in.

hawkeyexp, 2018-08-26:
@all,

please try out following combi:

`/etc/pulse/daemon.conf`

```
# CSNG params
default-sample-rate = 48000
alternate-sample-rate = 16000
resample-method = soxr-hq
default-fragments = 8
default-fragment-size-msec = 5
flat-volumes = no

default-sample-format = float32le
enable-remixing = yes
enable-lfe-remixing = yes
high-priority = yes
nice-level = -11
realtime-scheduling = yes
rlimit-rtprio = 9
```

`/etc/pulse/default.pa`

from:
`load-module module-udev-detect tshed=0`
to:
`load-module module-udev-detect tshed=yes`

Report back...

dreamvictim, 2018-08-26:
A definite improvement over the default settings. The Bass seems really loud compared to the settings I was using but sound quality is good. This was tested using my Hifiberry we need someone to test using the pi audio.

mishu28nmv, 2018-08-27:
I want to test this with RPi on board audio, but I have no such 'etc' folder. Only crankshaft and overlays

mishu28nmv, 2018-08-27:
https://imgur.com/eZTICfP

Maju3, 2018-08-27:
Yes on Windows Systems you only see the boot Partition of the sd Card.
https://www.paragon-software.com/home/linuxfs-windows/#

This is a Windows Software which Shows all partitions of the sd card

mishu28nmv, 2018-08-27:
Ok thanks. Will download and try.

Maju3, 2018-08-27:
No problem.
You will see the sd Card Partition "rootfs" then.
This is the place to edit;)

mishu28nmv, 2018-08-27:
Is 'tshed' or 'tsc**h**ed' correct? Here is 'tshed' and in my file I have 'ts**c**hed'.

hawkeyexp, 2018-08-27:
soppy it is a typo :-(

mishu28nmv, 2018-08-27:
ok. thank you very much!

Maju3, 2018-08-27:
Just tested the new soundsettings.
Tested with my Allo Boss dac and to me it sounds like a huge improvement.
More bass and the music sounds more rich of details now.

Maju3, 2018-08-27:
For People who use the OnBoard Audio:
You can improve the OnBoard Sound by adding this line in boot/config.txt:

audio_pwm_mode=2

This enables the new OnBoard Audio Driver.

mishu28nmv, 2018-08-27:
What do we use with onboard audio? Qt or rt. Sorry if this a dumb question but I don't know the meaning of those 2 things.

dreamvictim, 2018-08-27:
Try them both and see what works best for you. RT works best for me, The voice on QT goes out of sync or glitches for a lot of people.

Maju3, 2018-08-27:
Yes, RT audio works better for me,too.

hawkeyexp, 2018-08-29:
Some tweaks will be added on next release - closing for now.

closed 2018-08-29 by hawkeyexp