opencardev/crankshaft #209: [CSNG] Crankshaft + KODI

LoganFanGer, 2018-08-25:
This really is a good sense to integrate Kodi. Of course, I immediately tried it. As a skin I naturally favor CarPC-xTouch (3.1.1) from RasPiCarprojekt.de: http://raspicarprojekt.de/showthread.php?tid=790

So far so good. Function is there, but I can not save changes to the skin (color, arrangement of elements). After a restart, the skin is back to default. Where do I still have write permissions? Apparently, the changes are not written to guisettings.xml.

Of course, this is primarily about CS, but maybe someone has a tip. Kodi is just an additional construction site ;-)

chrisfromwa, 2018-08-25:
OK,  that is pretty cool.  I can't read any of the thread though since it's in German.  :(
You got this all working with Crankshaft?  Do you have the FM radio feature working?  This is something I really want to be able to have,  to totally replace the headunit in my car.

inspector71, 2018-08-30:
Has anyone looked into interacting with Kodi via it's JSON API, uPnP support, or even using Yatse as a 'bridge'.

I imagine everyone's thinking first about running Kodi on their Crankshaft hardware. However, I think it might also be worth investigating running Kodi on our phones if Kodi could become an Android Auto compatible app. The overhead of serving high FPS through Android Auto's so-called projection might be too much, especially over WiFi, or not. Audio should be fine. Anyone seen Android Auto compatible video apps? Hmmmm. 

Might be limited by perceptions of safety issues when viewing video on head units. The Android Auto Getting Started Developer guide https://developer.android.com/training/auto/ seems to focus on just audio and messaging.

hawkeyexp, 2018-08-30:
Kodi will never be official supported by crankshaft. I'm sorry for that but i fully agree with Huan and Google - Video playback on the headunit is a safety risk for you and also for OTHERS. That's the clear point why it won't be integrated and issues about kodi are closed.

closed 2018-08-30 by hawkeyexp

inspector71, 2018-08-30:
Ouch 😀

Very finite!

Makes sense in that safety has to be an absolute at some level.

I kinda feel like I've been slammed for even considering the idea though.
When I actually did mention the safety concerns. I can see how even the
pause when moving isn't a solution because cars are often stationary but
drivers still need to be conscious and spacially aware.

Shame for audio though. But if Kodi can't be limited to just the audio
library, as I imagine it cannot, then it's safety first for sure.

I personally am very annoyed that the laws in my location allow phones,
navigation, taxi company meters to all be suction cupped to windscreens
just beside the driver's field of vision, but within peripheral vision. I
feel that's equally as dangerous. I've experimented with my phone running
Torque and blutack under my dash binnacle, with Bluetooth to OBD. But it's
often times just too distracting. Touch screens and their swiping etc are
too fiddly to use distinctly.

This is why I'm so into Crankshaft.

Steering wheel controls for eyes-up input is the only option I've been
interested in and one big reason why I don't yet have a functional
replacement for my OEM setup.

chrisfromwa, 2018-08-30:
I personally couldn't care any less about video,  but I was hoping to use KODI for AM/FM radio or maybe other audio sources.   But I understand the issue with video and why it's not going to be supported in Crankshaft.

Is there any other software out there,  that people recommend for radio?  I really want to use this as a head-unit replacement and I want to have some functionality in the unit when my phone is not connected.  

Does anyone know of anything,  or willing to experiment to get something working?  I'm not sure what hardware/software to be looking for.

hawkeyexp, 2018-08-30:
@inspector71 don't feel slammed - i can fully understand the motivation and i love kodi, too.

inspector71, 2018-08-30:
I've an RTL SDR unit I'm hoping will provide an option for radio. Don't
know how the software side will work out but that's where I was going for a
radio option.

https://www.rtl-sdr.com/

inspector71, 2018-08-31:
@hawkeyexp 

No worries. Safety first.

Not to be unfairly subversive but if people want Kodi integration enough, they can get it one way or another. You don't have to support it. You're the one who feels the responsibility for people's safety and that's hugely respected by me. It's good to have ethics and integrity in this project!

bnevets27, 2018-09-02:
I've never understand hampering something in the name of safety. Anyone can stick a phone or tablet on a dash and watch movies all day long. I would venture to guess most if not 100% of the users of CS want something that works easily and is safe.  If they didn't care for either of those things they would just stick a phone or tablet on their dash, 

The use of watching video on the screen when parked would be nice if to either kill time waiting or to entertain a passenger. Not frequent scenario but valid nonetheless.

I don't particularly care if kodi is included. If it was I would likely use it. And my current install of just a tablet lets me play video via plex (or kodi if I choose). Having the ability to play video is nice and not using it while driving is just smart.

hawkeyexp, 2018-09-02:
@bnevets27  There is nothing active done to prevent from modify your system or installing software. You can change your system yourself. But please respect the decision we have done. Thx.

bnevets27, 2018-09-02:
I understand, I'm glad it isn't being blocked, it nice working with kind of open source software.

And the developers(you) are the ones to make the decisions. 

I was just making the point that not integrating something purely out of a safety standpoint doesn't really prevent people from doing dumb things. 

This was in reference to:
"Video playback on the headunit is a safety risk for you and also for OTHERS. That's the clear point why it won't be integrated"

But like I said your software so it's your decision. Even if I don't agree with the reason behind it. 

I hope this, and the previous didn't come off as rude or harsh. I'm grateful to the development done on this project.