opencardev/crankshaft #210: [DOC] WiFi connection

inspector71, 2018-08-26:
Hi

I'd like to copy / paste and write a wiki document on how to connect with WiFi based on this thread:

https://www.reddit.com/r/crankshaft/comments/99py4c/wireless_connection/

Specifically, @FrankLaVerne's post here:

https://www.reddit.com/r/crankshaft/comments/99py4c/wireless_connection/e4t5g65

I would then modify the [Form factor section](https://github.com/opencardev/crankshaft/wiki/Getting-started-with-Crankshaft#form-factor) to include a note about WiFi connectivity to give context to the "- most simply via USB cable -" text.

I've cloned the WiKi repository and can use git on the command line but have yet to interact with a project based in github so could use a little hand there just to establish the editing workflow. I must admit, my style tends to be very iterative. In other words, I make lots of mistakes and often only notice them after publication :) Is anyone aware of a good offline markdown preview tool?

hawkeyexp, 2018-08-26:
the wiki is open again - have fun :-)

Codehhh, 2018-08-27:
@inspector71 Visual Studio Code has a pretty solid markdown editor built in, with some extensions to enhance support.

[Documentation](https://code.visualstudio.com/docs/languages/markdown)