opencardev/crankshaft #211: (CSNG) Ignition pin

Maju3, 2018-08-27:
Okay for now you can set a ignition pin in cs settings which causes a shutdown when this pin is HIGH.
Now i wonder if it is possible to include a BOOT script to make the pi booting when the ignition pin is LOW?

That way you can get a very simple power management by use of a 5 pin relay.
The problem is that motorstarts of the car makes the relay loosing power.
About this you need a BOOT script what works ONE TIME only.
It needs to work like this:
You turn your cars ignition ON and the relay sets the gpio ignition pin to LOW.
Pi boots and directly stops the boot script.
And when the pi shutsdown then the boot script will be reactivated.
That is needed to avoid that the boot script fires up again on motor cranks,you know.

hawkeyexp, 2018-08-27:
That's not correct - The pin must be LOW for delay time to init shutdown.
Rest must be done by the psu cause if pi is down it is fully stopped.

closed 2018-08-27 by hawkeyexp

Maju3, 2018-08-27:
Yes the CS ignition pin must be LOW to init a shutdown.
Thanks for the correction :)
Anyway you can boot the pi even it is fully stoped after shutdown by Setting pin5 (gpio3) to LOW.
This will init a boot.
The Problem About gpio3 is just that most dac hats use this pin.