opencardev/crankshaft #212: (Csng) BT pairing not possible on PRE 7

Maju3, 2018-08-27:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

I just tried to pair my phone with pre 7.
Not possible!

Swaped my sd card with pre 6 installed and bt pairing works perfectly.

hawkeyexp, 2018-08-27:
Parings will be stored during updates - repairing is usually not needed.

Delete pairing on phone, remove all paired devices inside settings -> wireless.

Pair again.

Note: pairing is only available on first 120 secs. - after this time press the bluetooth logo on mainscreen to enable pairing again.

Maju3, 2018-08-27:
I have formated my sd card and did a clean install of pre7.
No update.
When i start the pi i have 120 sec to make it pairable.
After the 120 sec I need to click at the bt icon in homescreen to make it pairable.

Anyway my phone can not see CS as pairable bt device.

Another sd card with pre6 on it makes no problems.

closed 2018-09-04 by hawkeyexp