opencardev/crankshaft #213: AudioInjector Octo support

inspector71, 2018-08-30:
Hello!

As discussed over on Reddit, would very much appreciate support for the AudioInjector Octo DAC if it's not too much burden on developer time.

Hoping to help repay effort with testing, wiki help, Reddit sub support. Will add more resources to this issue ASAP.

hawkeyexp, 2018-08-30:
Try this build of csmt:

[csmt](https://raw.githubusercontent.com/opencardev/prebuilts/14e82ae38565b1a727782a164d5126f32614db2b/csmt/crankshaft)

Replace` /usr/local/bin/crankshaft` with it.

`crankshaft audio audioinjector octo` should do an initial setup for the octo (first as plain stereo).

Please check if it is available in settings -> pulseaudio input/output.

Create a debuglog and add it to this issue.

inspector71, 2018-09-08:
Thanks for this. Some initial success! There's octo lines in the pulseaudio output and input, however only the input was selected by default.

inspector71, 2018-09-11:
Wish I could say yes to stereo. Haven't quite got that amp yet.

Maybe I'm spending too much time on CS, not enough time implementing CS? 😀 

Will certainly keep you posted in time!