opencardev/crankshaft #214: Screen flickering since NG PRE6

crstmkt, 2018-08-31:
- What phone do you have?
Nokia 6.1 (aka Nokia 6 2018)

- What OS version?
Android 8.1.0 (August 2018)

- What crankshaft version?
PRE6 (also PRE7)

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
I updated my installation of PRE5-1 to PRE6 using the Auto Update Feature. Since then I experienced random screen flickers (Screen goes black for half a second or so)

- Please provide any further information that you might find helpful if available.
After experiencing these flickers, I formatted the SD card and installed PRE7 from scratch - same symptoms.

hawkeyexp, 2018-08-31:
There was nothing changed which can cause this. Flickering is caused in most cases by to low input voltage bad cables etc.
Which display ?
Did you get the yellow flash top right ?

crstmkt, 2018-09-01:
In order to get rid of this flickering, I made a clean install of PRE5-1 but als experienced that flickering, so I went back more to PRE4. Using PRE4 I experienced that flickering not yet (tested on a ride of about two hours).  Starting with PRE5-1 that flickering began (multiple times in a minute).

I did not get the yellow flah - also my screen and the pi are powered seperately so I just pass the Data-signal from the pi to the screen.
I am using this screen: https://www.chalk-elec.com/?page_id=1283#!/7-black-frame-universal-HDMI-LCD-with-capacitive-multi-touch/p/21750201/category=3094861

hawkeyexp, 2018-09-01:
i have no idea why it will happens sorry.
One thing you can try if it has some effect - disable the cpu freq scaling of the system.
`sudo systemctl disable cpufrequtils`

Omnicrash, 2018-09-10:
@crstmkt Can you measure the voltage (5v pins)? I've had this happen too to a lesser extend, with the screen flickering every ~1sec, and in my case it was because I wasn't quite getting 5V.

hawkeyexp, 2018-09-11:
I could identify some direct impact between cpu governor (which is set to performance) and screen flickers on the original pi display. In next build gov is set to conservative which limits this effect and is enough for the system workload. But generally a stable power source is essential - not only stable 5V - also the stabilization done in the psu is essential!

hawkeyexp, 2018-09-21:
@crstmkt can you check adding this to config.txt if your flicker effect is still present:

`avoid_pwm_pll=1`
`disable_audio_dither=1`

i could reproduce the effect and this killed it.

hawkeyexp, 2018-09-23:
Than sorry no other idea

crstmkt, 2018-09-26:
Currently I'm back on PRE4. Will prepare a SD Card with PRE10 in the next days and try to produce some debug logs / error logs. Maybe you can investigate them and find out whats happening

closed 2018-10-10 by hawkeyexp

crstmkt, 2018-10-14:
[debug.zip](https://github.com/opencardev/crankshaft/files/2476177/debug.zip)
Sorry for the wait. I was on vacation the last few days and then totally forgot this. I attached the debug.zip from BOOT drive. I also recognized, that with PRE12 These flickering does only occur when my phone is connected to the Pi. Maybe this is a problem with the cable? But if so, why doesn't this occur in PRE4?