opencardev/crankshaft #215: Microphone not working since Pre 4

peb23, 2018-09-02:
Since Pre 4 the microphone does not work for me after activating google assistant, however if it reads a message to me and asks me what to do next it does record audio. I am using the recommended microphone. The capture vol. is set to 90%.

- What phone do you have?
Galaxy S8
- What OS version?
Oreo 8.1
- What crankshaft version?
NG Pre8 2018-08-31

hawkeyexp, 2018-09-02:
Please add a debug log. Thx

peb23, 2018-09-05:
Sorry for the late reply. Ive uploaded the debug file. In the meantime i tried to do a clean install of the latest release (pre8-1) but the issue remains.

hawkeyexp, 2018-09-21:
Should work - same usb mic (chip) used and working fine.

closed 2018-09-21 by hawkeyexp

peb23, 2018-09-21:
Unfortunately it still isn't. But I think i know what the issue is. I'm using a USB Soundcard with a 3,5mm audio output and a microphone input, as well as the USB microphone. When I unplug the USB soundcard the microphone works. It seems that when I select the USB soundcard as my audio output it blocks the separate USB microphone even though it is selected as the input device

hawkeyexp, 2018-09-21:
I had a similar effect in another part with a soundblaster usb soundcard which was dominant and killed the selected audio device.
I'm using a hifiberry dac plus standard with phone jack as audio output which works perfectly without any problems.