opencardev/crankshaft #219: [CSNG] Enhanced behaviour of ignition shutdown

rubenverhoef, 2018-09-05:
scenario:
I'm at the petrol station with my phone connected to RPI, AA is running. I shutdown my car and refilling it, after it i'm locking my car and going to pay. When i come back (more than 1 minute after shutting down my car ofcource) my rpi is also in a off state.

What i want: 
The RPI should not turning off when my phone is connected (and AA is running). So when i'm refilling my car i'm not having to wait for the rpi to boot and put my music back on.

How to implement:
The ignition timer needs to check the phone connected state
or
The phone disconnected timer needs to check if ignition is still on (if so not turning off the system). This because when i'm not using AA (so my phone is not connected) i want to have the RPI screen on for the time and BT streaming or whatever.

Maju3, 2018-09-05:
I think it is not good to make the phone disconnect timer watching the ignition timer.
Or vice versa.
Cause then it is a MUST to use a relay or anything else to read the ignition state.

So like it is for now it is more flexible i think.
Cause now you can use the phone disconnect OR the ignition timer.
BTW you can even disable that timer what you do not need ;)

Maju3, 2018-09-05:
@rubenverhoef
Why not just disable the phone connect timer and set a time delay for the IGNITION timer?

Set ignition timer to lets say 10 minutes and you are fine.
And the PI will not shutdown when your phone is NOT connected(BT streaming).

rubenverhoef, 2018-09-05:
@Maju3 If the ignition pin is defined there is no problem to use that state.
The ignition timer can now be set to 1m max and when it would be possible to set to 10m then when i park my car it is on for 10m which will seek a lot attention.

ofcourse this will be an extra option/setting (checkbox to enable)

Maju3, 2018-09-05:
Okay.
I did not know that the ignition timer is limited to 1minute.

Maju3, 2018-09-05:
Anyway i would just suggest to increase the ignition timer to 15 minutes max.
This time should be enough for short stops (gas station).

@rubenverhoef
Remember the SCREEN OFF timer.
It turns off your screen to prevent your car from attention.

Maju3, 2018-09-05:
Yes i agree with you.
Stop ignition timer WHILE phone is connected is a good solution.

closed 2018-09-05 by hawkeyexp

Maju3, 2018-09-08:
Hmmmm I think this idea was not as good as thought.
Now the ignition shutdown only works when your phone is disconnected.
I think this makes the ignition shutdown obsolete cause you simply can use the phone disconnect shutdown instead.

rubenverhoef, 2018-09-08:
No you can not. When using the phone disconnect timer it will also be triggert when just driving around and not want to use AA and you want to see the RPI home screen

Maju3, 2018-09-08:
You are right.