opencardev/crankshaft #220: hifiberry digi not working on pre8

Hisma, 2018-09-06:
Hifhi berry digi does not produce sound on pre8.  

Troubleshooting steps taken - 

- verified alsamixer was not muted
- verified alsamixer was selecting the right device with the right output
- replaced files in /etc/pulse directory with older versions provided by hawkeyexp.  Did not fix
- commented out various cs-specific configuration changes in  pulse/daemon.conf file.  Made no difference.  Problem is likely not related to daemon.conf configuration.  
- was able to produce sound in dev mode WITH OPENAUTO NOT RUNNING (shell only), using an mp3 file and mplayer.  Mplayer reverted to the alsa driver to play the sound.  It tried the pulse audio driver first but it failed.

Further Useful information - 

- hifiberry digi is shown as an input and output option in the "audio" settings, and everything appears to be fine in the audio configuration, but sound is completely muted.
- when playing an mp3 with mplayer in dev mode/shell only, an error is given "pulse audio driver failed - permission denied".  Mplayer then reverts to the alsa driver, and that is when music is able to be played.  This is probably due to the fact that pulse audio driver is not manually loaded so it reverts to alsa.
- when trying to play an mp3 with mplayer in dev mode and OpenAuto running, an error is given "audio device got stuck!" and it repeats continuously until interrupted.  Sound does not play.  I believe this is because pulse audio driver is running in the foreground and therefore it cannot revert to alsa.  

The problem clearly seems to be related to the pulse audio driver.  I do not know much about how linux processes audio, but it seems that the alsa driver is able to produce sound without issue, whereas pulse audio is not working.  Looking at hifiberry documentation, pulse audio is NOT required for the device to work - 
https://support.hifiberry.com/hc/en-us/articles/205377651-Configuring-Linux-4-x-or-higher

It should be able to work with alsa alone.  This was confirmed with mplayer successfully playing music using the alsa driver only (after failing to use the pulse audio driver).  There is no official configuration available on the hifiberry website to configure the hifiberry for use with pulse audio.  

I do not know what else to try at this point.  I think it would be worth seeing what changes were made that would affect audio output between pre3 and pre8.   Perhaps bt modules or some other configuration changes outside of daemon.conf are the cause.

hawkeyexp, 2018-09-06:
Some audio restructuring is in progress currently so please wait...

But a clear point: system without pulseaudio is not possible. Pulseaudio is always needed to allow openauto to output audio to the hardware.

Hisma, 2018-09-06:
Yeah I didn't say I understood the audio system very well :).  I just saw that I could get audio to work fine without it.  If that is the case and pulse audio was working for me in pre3 and got broke sometime after, then that is encouraging.  That means that it's capable of working fine on my system with the right settings.  I will stay on pre3 for now, and when an update is available with restructured audio I will give it another try.

Hisma, 2018-09-21:
ended up being my cheap clone.  it is best to use the official hifiberry parts and not cheap clones.  When I swapped for an official hifiberry digi sound worked perfectly.

Hisma, 2018-09-22:
this problem wasn't what I thought it was.  It was a problem with '/etc/pulse/system.pa' settings.  It affected both my hifiberry clone as well as my official hifiberry dac+ pro. 
sometime between pre3 and pre8, 
### Automatically suspend sinks/sources that become idle for too long
load-module module-suspend-on-idle

was removed from system.pa.  Re-adding that module to system.pa in the latest cs pre10 brought my sound back.  Why it works, I have no idea, but it fixed it.

commit 69b851cf (2018-09-22): [script] Add back pa suspend-on-idle module / Fix #220
closed 2018-09-27 by hawkeyexp