opencardev/crankshaft #223: Crankshaft not properly booting since pre7

mboettger99, 2018-09-08:
I´m using a Raspberry Pi Zero W in combination with a "no name"  7" touch display.
Crankshaft was working properly all the time until pre7.

In pre7 it stopped booting properly at all... (it showed he three dots at boot and than rebooted)

In pre8 and pre8-1 its booting, but unable to start OpenAuto in EGL-Mode. OpenAuto keeps crashing all the time

hawkeyexp, 2018-09-08:
The Pi's based on armv6 are not supported. The project base is a rpi 3. During updating and optimizing qt5 got a recompile to be better fit to destination architecture and is only running on pi2 and 3. Sorry.

closed 2018-09-08 by hawkeyexp

hawkeyexp, 2018-09-09:
if you have a pi2 laying arround it will also work but pi3/3b+ has more power

Hein-NonesensE, 2018-11-26:
> 
> 
> The Pi's based on armv6 are not supported. The project base is a rpi 3. During updating and optimizing qt5 got a recompile to be better fit to destination architecture and is only running on pi2 and 3. Sorry.

Could you please post this information more prominently? E.g. in the wiki FAQ, getting started, installling pages? I ran into the same issue today with a Pi 1 B+ and was just in the process of posting the issue when I found the answer here.

hawkeyexp, 2018-11-27:
Done