opencardev/crankshaft #224: String corrections in the Settings app

inspector71, 2018-09-10:
Hi

I'd like to contribute by improving any typos I see. As I just edited the Settings Menu wiki page, I noticed a few things I could correct. I tried cloning the repo and `git grep`  but did not find some strings I was expecting.

Should strings in the Settings app be editable from the cloned app?

Thanks

hawkeyexp, 2018-09-10:
? I could see some changes while resync wiki repo.

to bring up your local clone do a `git pull` inside which pulls latest changes and merge it to your local clone.

hawkeyexp, 2018-09-10:
PS: the changes looks good :-) thx

inspector71, 2018-09-10:
Whoops! I wasn't talking about the wiki repo :) 

I've gone back to editing via the web for that, for now.

I 'm referring to the actual Crankshaft repo. I thought I'd try and help out with small tweaks like ...

> Use big clock in classic (only if show clock enable)

instead of the current ...

> Use big clock in classic (only if show clock enable

But I cannot find that string in the source code.

hawkeyexp, 2018-09-11:
Ups sorry misunderstanding :-)

It's loacted in the settings ui -> https://github.com/opencardev/openauto/blob/crankshaft-ng/src/autoapp/UI/settingswindow.ui (line 364 in current file)

inspector71, 2018-09-11:
Oh wow, ok. No wonder I didn't find it. Wrong project! 😀 Thanks!

closed 2018-09-11 by inspector71