opencardev/crankshaft #225: Invert ignition pin behavior using config

Omnicrash, 2018-09-10:
I'm using NG Pre8-1, and it seems the ignition pin needs to be HIGH to initiate the shutdown. This is also indicated by the configuration file:

> # This pin must be low to keep system running. If high for > IGNITION-DALAY (seconds)
> # system will do a shutdown

This condracticts what is discussed in issue #211, where it is said that the pin needs to be LOW to trigger shutdown.

However, in my case it would be desirable to have that inverted, where HIGH would keep the system running, and LOW would indeed trigger the shutdown.

Can an option be added to configure this behavior?

Omnicrash, 2018-09-10:
I've implemented this using an extra parameter in the crankshaft_env: IGNITION_INVERT. Default is set to 0 to maintain current default behavior.

This option is not yet configurable via the UI.

hawkeyexp, 2018-09-11:
Your are partitial correct - the gpio predefined state was not set correctly caused by moving pin from general gpio controlled to standalone working. - fix is on it's way...

hawkeyexp, 2018-09-11:
Fixed and working - the gpio is predefined to high - if set to low all is fine and system keeps running - if high (open) shutdown will be done afer delay time...
Needs a doc ...

closed 2018-09-11 by hawkeyexp

Omnicrash, 2018-09-11:
Hi @hawkeyexp,
That doesn't actually fix/implement what I wanted to do, which is have the option to invert this behavior, where shutdown would happen when the pin is low for the duration of the delay timer. In my pull request I've added an extra parameter that makes possible, will you consider merging it? https://github.com/opencardev/crankshaft/pull/226 ?

The only changes are the introduciton of the parameter in crankshaft_default_env:
`IGNITION_INVERT=0`

And in service_gpiotrigger where it's value is used:
`if [ $IGNITION_GPIO -ne $IGNITION_INVERT ] ; then`

I've tested this on my setup, and it seems to work correctly.

hawkeyexp, 2018-09-11:
I know but invert makes no sense cause it's not failsave. This way for example a broken connection will also shutdown the system while gpio is not longer triggered low. If you need an inverted behaviour you must do it external - or rework it for your personal usecase - sorry.

Omnicrash, 2018-09-11:
@hawkeyexp That's why I left the default value at 0 - so if you need it, you need to first configure it and make sure your hardware is in order, so the connection should never be broken. If the user does set the parameter to 1, or if they remove it, it will still work as it did before.

I don't think users should be prevented from their own mistakes by not providing the option, and this will make it easy to use the NC connection of the relay to trigger shutdown.