opencardev/crankshaft #227: [CSNG] Pre 8-1 menu visible behind custom apps

dreamvictim, 2018-09-15:
When launching apps via the new custom buttons the CSNG menu is visible in the background. I know you've fixed this on previous builds. Would it be possible to black the screen on a custom app launch please?

hawkeyexp, 2018-09-16:
which app you are starting ? A general solution can't be done - i had only added an experimental solution for kodi

dreamvictim, 2018-09-16:
Kodi is one of the apps I'm using. Your experimental solution worked well. How can I implement it in the newer menu?

hawkeyexp, 2018-09-21:
Delayed for later build

hawkeyexp, 2018-09-29:
Some response to this:
In the past for kodi i have done it with a check which tests if kodi is present in the process list every second if it was installed and present on filesystem. The custom commands make this more complex cause it can be everything and possible hard to check.
My idea is to clear the screen if a custom button is pushed. But this requires a addition action which must be done from custom command to set the signal to openauto that process has exited. For example touch custom button sets screen blank and creates /tmp/customtask - until this file is present screen stays blank. If file removed (must be done from custom task) gui comes back - a possible solution ?

hawkeyexp, 2018-09-29:
Solution done for next build.
Creating /tmp/blackscreen cleans screen from controls and sets bg to black - removing reverts to normal mode.

closed 2018-09-29 by hawkeyexp