opencardev/crankshaft #228: (CSNG) Ignition pin delay timer

Maju3, 2018-09-16:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

Is it possible to increase the delay to 5 minutes?
And if possible in 0.5 minute steps?
I ask cause in my case the perfect shutdown delay would be 4:30 minutes.

rubenverhoef, 2018-09-16:
Why would you want that? Give us an scenario so we can understand your issue better.

hawkeyexp, 2018-09-16:
The idea of ignition shutdown is to trigger shutdown with ignition for example leaving car. With this base shutdown should be nearly instant - the delay time is designed to skip possible restart engine if needed without shutting down system. For longer cases the normal shutdown should be used.

Maju3, 2018-09-16:
The point i ask for this is because with gpio ignition pin you have the possibility to shut down the pi with your cars ignition.
But you can not boot the pi just by turning on ignition.
Lets say you power the pi from INSTANT power of your car(battery) and you connect ignition power from car to the pi(gpio ign).
Then after the shutdown is done you need to pull the powercable or bridge the RUN pins to boot the pi.

In my case i use a M3ATX psu.
Normally this psu is made for atx pc boards but it works with the pi.
On this psu i can set a hardoff after ignition is lost for 5 minutes.
Hardoff means it cuts the 5 volt to the pi completelly.
Exactly like pulling the powercable.

Now if i get a shutdown delay setting of 4:30 then after this delay the pi shuts down and 30 seconds later my M3ATX cuts the power completely.

This makes a great boot/shutdown solution by ignition key :)

Hisma, 2018-09-24:
@Maju3 I had this exact same scenario, except I wanted to extend the ignition time so that i can use the timer to trigger a shutdown script.  However after talking with @hawkeyexp he convinced me to use a script to trigger the shutdown timer instead.  It works, though in your case, shutdown timer is set in minutes, so you can set the shutdown time to 4 minutes in CS BASE, then modify startup.sh.  
Here is my script - 

```bash
#!/bin/bash

#This sample checks MY_GPIO which is predefined to high for low stage every second

# Define my gpio
IGN_PIN=23

# predefine gpio state (up = high)
sudo /usr/bin/gpio -g mode $IGN_PIN up

while true; do
    if [ $IGN_PIN -ne 0 ]; then
        IGN_GPIO=`gpio -g read $IGN_PIN`
        if [ $IGN_GPIO -ne 1 ] ; then
			if [ ! -f /tmp/ignition_on ]; then
                touch /tmp/ignition_on
				/usr/local/bin/crankshaft timers shutdown stop && /usr/local/bin/crankshaft display on
			fi
        else
			if [ -f /tmp/ignition_on ]; then
				rm /tmp/ignition_on
				/usr/local/bin/crankshaft timers shutdown start && /usr/local/bin/crankshaft display off
			fi
        fi
    fi
    sleep 1
done
```

When ignition gpio detects ignition off, it turns off the pi screen and starts the shutdown timer (which for me is set to 59 minutes).  If ignition is turned back on, the display is turned back on and the timer is stopped.  The script works well.  In my case, I have my pi connected to a physical delay off timer relay set to 1 hour.   

I do think at some point this feature should not require a script, as other units do this natively (most android-based head units have configurable delay times), and it's extremely useful to avoid the long boot time for short durations. But I think the point of the ignition pin is strictly for ignition delay, not shutdown delay.   In my opinion having a gpio triggered shutdown timer is much more practical than a "device not connected" triggered shutdown timer. Perhaps this can be considered for a future release.

Maju3, 2018-09-28:
Yes.
You already have a shutdown timer in cs base setting.
For now this timer is triggered by disconnecting phone.
So you just need a setting if you want to trigger this shutdown timer by phone connection OR ignition state.
Very simple :)

And yes, it does not matter if the pi shutsdown in 1 minute or in 2 hours after you leave your car.
Because the SCREEN backlight shuts off.
So there is no attention from thiefs:)