opencardev/crankshaft #231: Wi-Fi and mobile data

mishu28nmv, 2018-09-18:
I can connect with the wifi function but I don't have internet then, so no spotify or tunein. Waze is working but offline I think.
Tried to mark the "switch to mobile data when no internet from wifi" function, but still no internet.

- Samsung Galaxy S7

- Android 8.0.0

- Crankshaft Pre9

hawkeyexp, 2018-09-18:
Not every device can handle this dual connection. If you check reddit some devices are working fine with the current setup. There is no doc about the detailed config from google to identify which constellation allow both in a working case. The default scenario is the one or the other way - google is breaking this normal case.
I have the same effect but could identify a possible solution to get it working... But it's not really a cs problem...
I will keep the issue open but there is no way to solve it for all devices.

As i said in the wiki:

If it works, be happy. If not, there's nothing we can do, currently

mishu28nmv, 2018-09-18:
> I have the same effect but could identify a possible solution to get it working... 
I know that there is not a problem from CS:NG.
I have 2 option on the 'switch to mobile data' function: normal mode and agressive. I will try them both.
Can you share your solution? Thank you for everything!

internix14, 2018-09-18:
Well. Some phones need to be in developer mode. Or in xiaomi phones case. In developer settings you can activate an aggressive dual link. Try to see if Samsung phones got that

mishu28nmv, 2018-09-18:
Thanks all. I will try everything

mishu28nmv, 2018-09-18:
@hawkeyexp 
Do you mean like this:

# router
# dhcp-option=3,192.168.254.1 ?

hawkeyexp, 2018-09-18:
yes - comment it out...

mishu28nmv, 2018-09-18:
I did those 2 line with # symbol. Mount and see how it goes :)

hawkeyexp, 2018-09-21:
Have you tested ?

inspector71, 2018-09-22:
Pretty surprised by this. My ASUS phone has never needed any configuration changes to do this. I just toggle WiFi and Mobile data and it's not an issue. Guess I'm just lucky,

xHirscHx, 2018-09-25:
Just set this up its asking for password to conect to crankshaft.look on forms can't find default or I missed it. Ant help thx

dreamvictim, 2018-09-25:
@xHirscHx 1234567890

mishu28nmv, 2018-09-26:
> Have you tested ?

I think for me it's better just to plug the usb. I'm not using the wifi function anymore. Thanks for your support!!!

hawkeyexp, 2018-09-26:
Ok than i will close issue.

closed 2018-09-26 by hawkeyexp