opencardev/crankshaft #233: [CSNG] [Feature Request] WWAN Module with GNSS Sensors

cydia2020, 2018-09-19:
System Information
- Pixel XL
- Pie (PPR2.180905.006)
- Crankshaft: f6184eb

I think Linux has native support for some WWAN cards like those from Qualcomm and Huawei, now with the introduction of the Headunit Client, I think we can make the Pi share a Wi-Fi Access Point with the internet from an USB WWAN Module, which can make the life for a lot of people in the community a lot easier, as not everyone has a Nexus/Pixel device.

Data share cards for tablets and always connected PCs can be as cheap as $3 in some part of the world.

Implementing it should not be too hard as the drivers are already in place, we just need the correct USB Mode Switch, however, I don't know whether the Wi-Fi Device on the Pi can keep up with that rate of data transfer as we are already seeing lags all over the place, but we can always use a third-party USB Wi-Fi dongle and disable the internal Wi-Fi device.

Some useful (probably) products / links
https://www.aliexpress.com/item/600Mbps-5Ghz-2-4Ghz-USB-Wifi-Adapter-USB-Dual-Band-RTL8811AU-Wifi-Antenna-Dongle-LAN-Adapter/32900537041.html
https://www.aliexpress.com/item/Sierra-Gobi5000-EM7355-LTE-4G-WWAN-FRU-04X6038-NGFF-Wifi-Card-For-Lenovo-T440-X240-W540/32910047133.html
https://www.aliexpress.com/item/L850-GL-LT4210-FDD-LTE-TDD-LTE-4G-Card-4G-Module-SPS-917823-001-For-hp/32870801173.html

We need to use the non-OEM parts (not from HP or Lenovo) of these devices since the OEM modification can really screw things up.

Also, we can use the built in GNSS sensor on a lot of these cards to pass the GPS information from the Pi to the phone, decreasing the phone's stress, sometimes with Android Auto connected and Google Maps navigating, the phone is not even charging.

Hope to see this implemented soon

hawkeyexp, 2018-09-19:
Some things to think about:

- using a stick for umts/lte ok - but who wan'ts to bye two sim cards or possible pay for two conenctions? (country and provider dependend) ...
- GPS from pi - why? Usually mobiles are placed in front of the cockpit or near the front which allows gps without problems
- Performance: the pi has limited internal bus speed - all knows about some small bugs in audio etc. which sometimes caused by limits reached - routing all over the pi will results in some crazy problems i think and results in more problems than solutions.
- it can by much easier: if someone want's to use an umts/lte connection it is no problem to prepare the normal system mode to enable eth0 and you can connect an easy mobile umts/lte router.
The routing is included and working in dev mode - it was there for some tests since a while...

Only my assessment...

cydia2020, 2018-09-19:
As stated, data share SIM can be as cheap as $3 in some countries

E.G.
https://www.telstra.com.au/content/dam/tcom/personal/help/pdf/cis-personal/mobile/personal-critical-information-summary-data-share-sim-5.pdf

Okay if all the necessary files are in place I might look into this myself first as soon as my m.2 to USB adapter arrived.

Would you please share the method of enabling networking in normal mode while compiling?

Cheers

rubenverhoef, 2018-09-19:
why do you want to use M.2 to usb, i don't think that is working. You can better just using an LTE USB modem dongle. Wifi is already onboard and i don't think a usb wifi stick is giving better performance. And about GPS you can better use a GPS to uart module.

cydia2020, 2018-09-19:
Lmao I have a m.2 LTE card laying around, it's XHCI

smartroad, 2018-10-12:
"GPS from pi - why? Usually mobiles are placed in front of the cockpit or near the front which allows gps without problems"

mine isn't, it has to sit down in a small cubbyhole near to my gearstick. Mostly this isn't a problem but sometimes it can loose lock as it doesn't exactly have the best view of the sky. Not to mention in the UK the police seem to be getting quite OTT on what you can and can't do on a phone, best to have it out of the way, then there is no chance they can say you were using it.

Just another POV 😁

closed 2018-11-24 by cydia2020

danielsteiner, 2019-04-28:
Since I sometimes share my car with my dad, I was wondering if this would be possible - I know, that Gmaps / Waze propably are better, but I don't want my dad to have connect his phone since I am using AA in Wifi Mode and didn't take USB Cables into consideration - he isn't enough of a techie to configure Android Auto but he could use a navi sometimes.