opencardev/crankshaft #234: [Tracking issue] Wiki

inspector71, 2018-09-22:
Just thought I'd keep everyone informed of any wiki updates in a separate issue here. Might help to keep things syncronised or whatnot.

20180922: [Bluetooth page](https://github.com/opencardev/crankshaft/wiki/Bluetooth) re-written

hawkeyexp, 2018-09-22:
Good idea - thx...

DJFliX, 2019-01-19:
@inspector71 I was just told by @hawkeyexp that you are an active contributor to the wiki. I grouped and re-ordered the wiki menu entries but saw this issue and wanted to know if it makes sense to keep logging changes here. The last changes were logged about 3 months ago. If we don't keep logs of wiki changes here I guess this issue might as well be closed, right? :)

inspector71, 2019-01-20:
Hi DJFliX.

I think it makes sense to have a place where we can discuss wiki editing as randomly editing risks chaos.

I don't know how we could have that discussion space. Reddit, GitHub issues ... none seem quite right.

I've been somewhat lazy in not updating this issue as my motivation has been tainted by the annoying limits of GitHub Wiki and people in the Reddit often failing to acknowledge @hawkeyexp's efforts and just whinging about bugs (in alpha level software!).

I suspect it might make more sense to move documentation away from the heavily limited GitHub Wiki software. Perhaps to getcrankshaft.com

closed 2021-11-27 by github-actions[bot]