opencardev/crankshaft #235: Pre9 WiFi hotspot requires a WPA2-PSK?

inspector71, 2018-09-22:
Just looking to improve the wiki page on how to connect to Crankshaft from a phone over WiFi (instead of via USB cable). 

Fairly sure I did this in an earlier version and do not remember it requiring a pre-shared key (password).

Just went to try it again and I see the CRANKSHAFT-NG network in the WiFi networks list on my phone. When I tap it, a request for a pre-shared key appears.

Unsurprisingly, because I've no idea what the pre-shared key should be, when I tap the WiFi button on the home screen to tell Crankshaft which IP to look for the Android Auto _projection_ on, there's nothing listed.

I'm pretty sure the only time I hot this working, I was in Debugmode. So I guess I'm trying to test the GUI mode and hopefully update the wiki page with fuller instructions. But yeah, naturally I tried `crankshaft` as a guess for the pre-shared key (why don't the just call it a bloody password?!) but that didn't work. Short of brute force dictionary attacks, I guess we need to make it obvious what the password (nuts, ... "pre-shared key" ... LOL)  is. Either that or maybe I'm doing something wrong?

hawkeyexp, 2018-09-22:
a psk was always required and must be defined in crankshaft_env.sh. Possible your psk was stored in the past on phone side so it won't ask for it.

inspector71, 2018-09-22:
> 
> 
> a psk was always required and must be defined in crankshaft_env.sh. Possible your psk was stored in the past on phone side so it won't ask for it.

Of course! Makes sense. Thanks :)

closed 2018-09-22 by inspector71