opencardev/crankshaft #236: [Wiki] Future features and wiki progress

inspector71, 2018-09-23:
I know CS is in rapid development, which is great, however this can make some wiki work potentially redundant very fast. A good example might be the WiFi page. I'd like to try and get ahead of releases, if possible. In the case of WiFi, I suspect it might be best to allow users to set the hotspot password ("Pre-Shared Key") via Crankshaft Settings instead of a text file. Is that currently in progress? If not, I'll edit the WiFi page with a little more detail on where to set the password in the config file.

I guess I've managed to contribute ok so far without earlier Patreon access to new builds, or information on changes so I can get documentation ready. If that's the easiest way to keep proceeding, I guess that's ok. If it were possible for me to get more detail / earlier access though, that could help everyone.

hawkeyexp, 2018-09-23:
psk via settings was included in settings in the past but has broken some scripts during setup to crankshaft_env.sh so it was removed but should come back in the future :-)

Contact me via Hangouts - hawkeyexp@googlemail.com

inspector71, 2018-09-28:
No worries. I've tried to contact you via Hangouts but no luck so far though it's only been 7 hours :)

Do you think we should remove this:

> Caused by a damaged wiki we decided to close the open access to edit the wiki.
>
> The wiki will be updated from time to time. You are welcome to help bring the wiki up to date.
>
> In this case open an issue with [DOC] reference and put in your post.
>
> Thank you.

It's in the way of me possibly writing a more friendly home page.

hawkeyexp, 2018-09-28:
Sure...

HeisenbergMarc, 2018-10-04:
@hawkeyexp Where is the textfile where i can change the password?

inspector71, 2018-10-04:
I think you may be looking for:

`/boot/crankshaft/crankshaft_env.sh`

closed 2018-10-09 by inspector71