opencardev/crankshaft #237: Android auto dont want to start

krooozong, 2018-09-23:
Hey,

i am new at this forum and have a problem with rasperry pi 3 connected to a 7 inch touchscreen.
I just want to use it in my car with android auto. 

My problem is that the android auto dont want to start, as in the tutorials that i saw. 
My rasperry and screen starts normally but  i get not the screen of android auto. Just the wallpaper menu and the settings. So my question is: What do i have to change that my android auto starts automatically ???

I attach some pictures to see what screen i get the whole time
System: 2018-09-14- crankshaft-ng-F6184EB

[image]
[image]

tehniemer, 2018-09-23:
It doesn't look like you have a phone connected.

hawkeyexp, 2018-09-23:
@krooozong - this is the new gui with csng (pre-releases) so all fine.
You have to connect your phone and android auto should launch - it's only a projection which is working with phone connected no standalone solution. If you android auto didn't launch have a look into the wiki and create a debuglog.

krooozong, 2018-09-23:
okay thx. 

Thumps up

hawkeyexp, 2018-09-29:
seems like the issue can be closed...

closed 2018-09-29 by hawkeyexp