opencardev/crankshaft #239: Request: Throw away wake-up touchscreen click

beanot, 2018-09-28:
Nexus 6p on CS-NG Pre10 with official 7" touchscreen.

Currently screen goes blank on idle, and can be woken up by a click.
Problem is that the click gets processed, so no way to ensure that wakeup click won't result in unintended effect (e.g. switching audio track, changing nav, etc).
Any simple way to just throw away the click (after using it for wakeup of course), or perhaps on blanking change input mode so next click won't actually do anything beyond wakeup (e.g. have it act like hitting <SHIFT> key)?

hawkeyexp, 2018-09-28:
Yes i know - something will be released for this with next builds - currently tap on top center there is no control item

hawkeyexp, 2018-09-28:
Fix is done and will be available with next build. Was to late for pre11 sorry.

closed 2018-09-28 by hawkeyexp

beanot, 2018-09-29:
Thanks for quick response. Just installed pre11, looking forward to playing around in it tonight. Have a good weekend!

hawkeyexp, 2018-09-29:
Thx, you too.