opencardev/crankshaft #243: [CSNG] Automatic upgrade failure from pre10 to pre11

tehniemer, 2018-10-01:
I went about the upgrade process as I have since automatic upgrading was implemented, but this time the screen went blank in the middle of the process and never came back. The card is now write protected and I cannot flash the image manually or even format the card.

hawkeyexp, 2018-10-01:
sounds for me like youre card has died. Possible you can try to repartition the card if it fails it's clear and Ende oft life for the card. The flash process itselve is working fine and no changes was done.

hawkeyexp, 2018-10-01:
it can happen every time - no software prob

smartroad, 2018-10-04:
Not directly related to this but as a bit of moral support, I have a 128GB flash drive that died after only about a year and a half of use. Wasn't heavily used but still. Sometimes I think it is luck of the draw as to when a device will die.

hawkeyexp, 2018-10-07:
Closing cause not software related.

closed 2018-10-07 by hawkeyexp