opencardev/crankshaft #246: AudioInjector HAT is muted

ktrenx, 2018-10-05:
I am getting no audio from my AudioInjector Stereo HAT in CS, I have tried pre8-pre11. I have added 'dtoverlay=audioinjector-wm8731-audio' to config.txt and CS pre-11 now sees the card in options.
In other OS installs I have to unmute 'Output Mixer Hifi' in alsamixer, but doing so in CS dev mode doesn't save through a reboot, even if I follow it with 'sudo alsactl store'.
A user on Reddit suggests that it may be a PulseAudio issue and recommended contacting AudioInjector. I have tried asking on the AudioInjector forum, but they have not posted my question in their forum.

hawkeyexp, 2018-10-05:
- Please try first a non manual setup via settings cs-hw - select your dac and select save on exit and do a reboot. Try if output is working after this.
- Alsactl won't work cause it's a readonly filesystem in normal mode - use crankshaft audio alsactl save|restore instead which saves the alsastate to /boot/crankshaft/alsactl.state.
- An unmute command ist send to the card via autoapp_helper on start of openauto.
I know from another user his card is working. With latest builds.

hawkeyexp, 2018-10-06:
create a debuglog from this install and add it.

ktrenx, 2018-10-06:
[debug.zip](https://github.com/opencardev/crankshaft/files/2453267/debug.zip)

ktrenx, 2018-10-06:
you are correct, absolutely an oversight on my part. Everything is working great now, thank you for your help

closed 2018-10-06 by ktrenx

hawkeyexp, 2018-10-06:
you're welcome