opencardev/crankshaft #247: Update script enhancement

nkean, 2018-10-08:
When trying to update crankshaft using a usb drive, the script will throw an error if the zip file was copied to the usb using a computer running macOS.  The issue stems from the fact that macOS adds a hidden file for each "real" file on a drive that doesn't support extended file attributes.  These hidden files copy the name of the "real" file and append a ._ at the beginning of the file name.  In the file tree this causes the hidden file to appear first and thus breaks the update script when it looks for the first file ending in .zip.  Once I delete the hidden file, the update proceeds as expected and uses the correct zip file.  I'll take a look at the script and see if I can implement a fix, but thought I would post this incase anyone else had run across the same issue.

hawkeyexp, 2018-10-08:
Add the filenames which breaks the script - a skip of this should no prob.

hawkeyexp, 2018-10-08:
Should be fixed now - ignore is added based on your decription...

hawkeyexp, 2018-10-16:
done with next build

closed 2018-10-16 by hawkeyexp