opencardev/crankshaft #250: Wireless mode - possible to use phone hotspot?

yusijs, 2018-10-13:
Hi!

So currently we need to use the Pi as a hotspot, and connect our phone to it. However, this  can be a problem, since not everyone can use both wifi + mobile data (myself included). As far as OpenAuto goes, I believe the only rule is that the devices needs to be on the same network, which in turn should mean it's a possibility to use your phone as a hotspot, and in turn connect the Pi to it?

As far as I'm able to tell, the only blocker for this is that Wifi is disabled when not in dev mode - if so, is it possible to have Wifi enabled in client mode?

I have another interest in this as well - I recently wrote a simple internet radio application for myself that I wanted to install on the headunit that I can launch without my phone connected, but to do this, I need an internet connection. An idea I had was to install a usb 4g modem in my car, which would allow me to listen to the radio without my phone connected.

Thanks for your time!

parkerlreed, 2018-10-13:
Yes, phone as hotspot works just fine. Use the phone's address which is usually 192.168.42.1 (it'll be the default route that shows up on the Pi)

yusijs, 2018-10-13:
Question was more on how the connection to the phone was established - is it just to change the SSID and password in the env file now, and it'll work outside of dev mode?

inspector71, 2018-10-14:
@yusijs What phone are you using? Android version? I'm curious about which phones, perhaps OEMs, disable concurrent mobile and WiFi. I've owned Samsung and ASUS Android phones and IIRC, Samsung didn't block concurrent networking on my old Galaxy S2 and my current ASUS does not either.

hawkeyexp, 2018-10-14:
@inspector71 it's try and error until a device has android 9 which should official bring aa wifi to oem devices. Official supported are only google phones if firmware is lower than android 9 by google.

yusijs, 2018-10-17:
@inspector71 Huawei Mate 10 Pro. I have the setting available via developer menu (mobile data always on), but it's not used when the wifi doesn't have an internet connection.

inspector71, 2018-10-18:
Wow, that sucks. I hate it when software is crippled for no good reason.
That you can then intelligently opt out of that crippling but the system
doesn't function as it should with that override in place, really blows.

Is your phone essentially running it's 'native' Hauwei firmware? What
Android version are you running?

yusijs, 2018-10-18:
Yeah, stock Emui, android 8. I decided to get openauto pro, as it allows me to easily use my phone as the hotspot :)

inspector71, 2018-10-18:
OpenAuto Pro is an interesting one. I didn't know it existed and then found
out about it and wondered if Crankshaft would always be limited in
comparison. Then forgot about it again.

Are you using OpenAuto Pro within Crankshaft or is it a completely separate
distribution?

hawkeyexp, 2018-10-18:
I think we should close this issue cause it's more a discussion than an issue - you are welcome to use reddit for discussions.

closed 2018-10-18 by hawkeyexp

inspector71, 2018-10-18:
> 
> 
> @inspector71 it's try and error until a device has android 9 which should official bring aa wifi to oem devices. Official supported are only google phones if firmware is lower than android 9 by google.

@hawkeyexp yeah fair enough. Another way of saying "trial and error" is "crowdsourced knowledge" :) Would be great if all CS users could help us generate a list of devices that have, or do not have, this wifi / mobile data concurrency issue.

hawkeyexp, 2018-10-18:
@inspector71 for me a list feels useless cause 8.x is not official supported for aa wifi and it can change every time with every firmware update etc.

yusijs, 2018-10-18:
Theres also the hw support issue here somewhere - I wrote down that it worked w/o internet at least.

Also @inspector71, by opinionated I mean that the creator(s) is limiting parts of the system. That's not a bad thing - it often makes the system easier to work with for endusers, but the downside is that it's limiting those of us who want a more customized solution (by creating barriers to do certain things, such as this issue). 

Again - not a bad thing, just a matter of preference. :-)

hawkeyexp, 2018-10-18:
The system is NOT limited - you can work and edit on all parts - also the source is open - so nobody is  limited to fork it and customize it. And the not used wifi in the past was also no limit - the only supported devices was the pixel ones and this limit is starting to be reduced with android pie and was not caused by cs.

inspector71, 2018-10-19:
I expect 'limited' might be a reference to the safety approach you've taken
Hawkeye. That was a hard limit imposed on potential users. That's not too
day the reasoning for that limit was unreasonable. But let's not pretend
the limit isn't there, it is.

"Limited" is not necessarily a criticism. Just an observation. Software
developers make choices all the time. Those choices can become limits on
users. Forks are not necessarily a solution as they can take what momentum
a project has generated and spread the resources behind that momentum too
thinly. The original project suffers and the offshoot project never quite
takes off.

A lot of open source software is not actually very democratic. People get
drawn to it and contribute however they can but unless there's an open,
balanced system in place to debate and choose the direction the project
moves in, there's a lack of democratic process. A good example is OpenELEC
versus LibreELEC.