opencardev/crankshaft #254: Button for temporarily monitor turn off

LoganFanGer, 2018-10-17:
Is it possible to create a tile on the start page with a command, that turns off the monitor temporarily? I run the system at staedy plus and  for short stops, this function would be great! 

THX for your good work! System works great since many weeks.

LoganFanGer, 2018-10-18:
Thank you. I'll try my luck. But unfortunately I fail to create the config file. I also do not understand the manual "Custom Buttons". Do not get there......

hawkeyexp, 2018-10-18:
wha'ts the problem ? You can als use the csmt (crankshaft command) to create a dummy file to edit it

LoganFanGer, 2018-10-20:
I start in Dev-Mode and connect by Putty. Start Command "crankshaft buttons 1" to create a config file. 
Console says "Done" But neither in the folder crankshaft nor in crankshaft / custom a file appears for editing. What am I doing wrong?

[image]

LoganFanGer, 2018-10-20:
Argh...the "correct" command is "crankshaft buttons create 1". File "button_1" is now in folder /boot/crankshaft. 

A first success. Continue experimenting  ;-)

LoganFanGer, 2018-10-20:
2 step. I create an orange Button with the label "Monitor OFF". Works fine.

Monitor OFF#/your/command#255,127,036#

But what is the correct syntax to implement "crankshaft display on|off" in command section? And whats the right path?

Using command ""crankshaft display off" in console works, but i but I can not integrate that.

hawkeyexp, 2018-10-20:
You ned a simple if else condition - but as i noticed a start prob fro aa in another issue wait 1 or 2 days until the bugfix release will be available - i will extend the crankshaft command for you with a toggle.

LoganFanGer, 2018-10-21:
Super, thank you!

hawkeyexp, 2018-10-24:
The startup issues from pre13 which has stopped publishing is solved and a new release is on it's way - including a toggle command for you `crankshaft display toggle`

closed 2018-10-24 by hawkeyexp